zynthian/zynthian-ui #8: UI stucks after several program change from sequencer

ivan1071, 2017-07-27:
Hi, as described to jose on zynthian forum, i experienced a UI freeze (or unresponsive) if there are several program changes messages over a multichannel snapshot. 
Tried with 3 or 4 channels only.
This happens not immediately, but after a bunch of times. not always the same time. a bit difficult to reproduce with precision.. The fact is Zynthian continues to respond to midi messages from sequencer, but everything on UI is stuck. (hardware encoders included).
By now i'm trying to figure out if it's a fault of one of engines, (i used fluidsynth together with zynaddsubfx) or even sending CC changes makes the same freeze.

jofemodo, 2017-09-23:
Please, can you confirm if this bug is currently preset?

ivan1071, 2017-09-24:
I checked, and seems fixed now :), UI continues working

jofemodo, 2017-09-26:
OK! I close the Issue ;-)

jofemodo, 2017-09-26:
Probably fixed by 0a840fea998d13d48b4e109e6b3f038c4091699a

closed 2017-09-26 by jofemodo