zynthian/zynthian-ui #13: Several layers in the same MIDI channel: impossible to adjust volume for each layer independently

jofemodo, 2017-08-26:
Same happens for other standard MIDI controls. See:

https://discourse.zynthian.org/t/volume-zynthian-encoders-of-single-engine-affects-both-engines/943

mheidt, 2018-03-09:
I like to see this done.
I would be happy to have volume offsets per midi channel like transpose for the time being.
This is something I could look at.

jofemodo, 2018-03-13:
It's not so easy to do. I haven't found a good solution yet. Most of engines use CC7 for controlling the volume and if 2 layers share the same MIDI Channel, it will react to CC7 changing its volume to the same level. For avoiding this, we need to create dedicated MIDI ports for every layer in the MIDI-router and rewrite the autoconnector. Some MIDI messages should be intercepted and re-scaled for every layer depending of its relative/offset volume.

Meanwhile, some tricks can be used depending on the combination of engines.

I.e. these days i'm using Pianoteq layered with some kind of pad or soft-synth-string from ZynAddSubFX. I use the "expression" controller from ZynAddSubFX for adjusting the volume relative to Pianoteq.

mheidt, 2018-03-13:
What about my solution?
My scenario looks like the following:
Masterkeyboard, that controls two channels at the same time.
Using a Pedalboard and constantly sending VolumeControl messages.
I just need an absolute or relative offset for a certain channel.
Something like, use 10% of the incoming value.
Or -30. I don't know, which is best. This can't be done with the midi-filtering yet?
But anyways, it should be adjustable while playing via zynthian-ui

jofemodo, 2018-03-13:
Your proposed scheme is OK, but doesn't change nothing about what i told you. We still need an independent MIDI port for every layer sharing the same MIDI Channel... ;-)

mheidt, 2018-03-19:
I had the following vision. 
 - we had a famous politician who said, that those with visions should visit a doctor -

The midi-filtering could allow offset (absolute) and percentage (relative) value changes. Maybe something new beside ignore and transfer. change_abs, change_rel or the like
These filters who have values, should be shown in the context menu of the layer so that you can change the value in the zynthian-ui.
This means, that the filter rule needs a label.

jofemodo, 2018-03-19:
But it can't work, my dear @mheidt, without making the changes i told you.
With the current state of things, all the layers assigned to the same MIDI channel will receive exactly the same MIDI data-flow. Currently the MIDI filter system doesn't support "per-layer" rules because there is no mechanism for doing this. For implementing it, we have to add to the MIDI router (zyncoder) a dedicated MIDI-ports for each layer. I want to implement it, but it's a major change ...

Regards,

mheidt, 2018-03-19:
I know, but i don't see it connected. I strictly use one midi channel per layer and don't mix.
And the changes to the midi-filtering (change_abs, change_rel) are independent as well.
When you will have made the "per-layer" change, you want to change the filtering as well i guess.

We could first work on the midi-filtering and don't bind it to the zynthian-ui until the "per-layer" works

jofemodo, 2018-03-26:
I'm thinking about creating a "mixer" layer in the UI that allows to modify the ALSA mixer levels. It would be bond to Master Channel CCs. Do you think this could be a simpler solution to your "scenario"? ;-)

mheidt, 2018-03-26:
I don't quite understand your solution. Alsa volume is over all layers. How will you differentiate between layers? A new alsa filter at the end of the chain for each layer? Before it enters mod-ui?