zynthian/zynthian-ui #17: Zynaddsubfx xmz parameter format

ivan1071, 2017-10-23:
Hi, what about integrating ZASFX xmz parameter format ? xmz combines more layers and can manage complex patches in just one recall... 

or maybe could be imported and "transformed" in zynthian patch, (assuming zynthian loads only one instance of ZASFX even for several layers) if wanting a totally ZASFX based patch, obiously... ;)  _(ehm...This one is just a brainstorming garbage.. don't care about it __)

jofemodo, 2017-12-14:
Done in fe53f09..59a9355

jofemodo, 2017-12-14:
t’s not finished. It only will create the first layer in Zynthian UI, so if your XMZ have more than 1 part, you only will see 1 in Zynthian UI, although the parts will be created. I’ve to add the code that parse the created parts and create the corresponding layers in Zynthian UI.

jofemodo, 2018-03-29:
I create a new issue for implementing this.

closed 2018-03-29 by jofemodo