zynthian/zynthian-ui #27: Do you really want to save snapshot

mheidt, 2018-02-21:
Add a "Do you really want to save snapshot" screen before saving a snapshot.

I would like to do it myself. Seems to be an easy task to learn about the architecture

mheidt, 2018-02-22:
Is it necessary to build a new controller for this page?
zynthian_gui_snapshot_save 
?

jofemodo, 2018-02-22:
Although it seems a "silly task", it's not. Currently there is no "confirmation widget/screen" in the Zynthian UI, so a new GUI class must be implemented, although a simple one.

If you want to do it, i would help you with guidance and tips. Don't fear ;-)

+ Take a look to the zynthian_gui_info class, as it's the simple one and the starting point i would choose for the new class. 

+ The layout for the "confirmation widget" should be:
  + A text label with the confirmation request. It should be configured every time the confirmation widget is displayed.
  + A text label / button positioned bottom left => Cancel/No. Switch-2 events should be bond to it. 
  + A text label / button positioned bottom right => OK/Yes. Switch-4 events should be bond to it.

Depending of the kind of widget used for the answers, you should bind the right events.

+ Take a look to the zynthian_gui class. There are several functions that manage the hardware switches: "zynswitch_xxx". You should modify some of them for calling the new screen class when it's opened. Probably, some deeper modification is needed because the lack of a good event routing system (my fault!!)... so the best is we work together in this development: I will improve the zynthian_gui while you work in the new "widget/screen class", layout, etc.

Regards,

jofemodo, 2018-02-22:
I created the branch gh27-confirmation-modal for this development.

mheidt, 2018-02-22:
ok, i committed the first baby step.
Have a look at save_snapshot(self): in zynthian_gui
I would like to be able to call show_screen with an if and need to know the return value.
You planned to use this confirmation widget for more than saving the snapshot, if I read correctly between the lines?

mheidt, 2018-02-22:
Btw, I would ask that early bc in the case you pressed the snapshot button too long although you wanted to load a snapshot, you don't want to browse through the folders and get asked at the final step.

mheidt, 2018-02-26:
@jofemodo 
if event[0]=='S':
				self.zynswitch_short(event[1])
			elif event[0]=='B':
				self.zynswitch_bold(event[1])
			elif event[0]=='L':
				self.zynswitch_long([1])
			elif event[0]=='X':
				self.zynswitch_X(event[1])
			elif event[0]=='Y':
				self.zynswitch_Y(event[1])

long...should this be event[1] ?

mheidt, 2018-02-26:
you can take a look. I fixed the event[1] mentioned above

mheidt, 2018-03-21:
I have to test the latest commit yet.
It was a rebase and I  am not sure whether I did everything right

jofemodo, 2018-03-21:
Hi @mheidt !
You are right. There is a typo there ... i've fixed in master.

jofemodo, 2018-03-21:
I synced the gh27-confirmation-modal branch with master for avoiding conflicts.

mheidt, 2018-03-21:
Does it work?

jofemodo, 2018-03-21:
I didn't try yet. I didn't merge gh27-confirmation-modal into master, but master into gh27-confirmation-modal ;-)

mheidt, 2018-03-29:
Did you try now?

jofemodo, 2018-04-03:
What should i try? I've checked out the branch, but no confirmation is shown when saving a snapshot.

mheidt, 2018-04-04:
the confirmation screen is shown, when you bold press L/S.
Not in the end of the chain.
If it doesn't work, i have to rebase and check again.

jofemodo, 2018-05-08:
OK! It works for me too. I will extend your code for having reusable "confirmation dialog" ...

Thanks a lot!

jofemodo, 2018-05-08:
OK! I see you already make the task. Great!!

jofemodo, 2018-05-30:
OK! It's merged with master now ;-)
319f088..5ceba34