zynthian/zynthian-ui #32: No way to run as dummy

guysoft, 2018-04-09:
Hey,
I want a way to run and test zynthian-ui on a raspberrypi without a sound card.
The problem is that it will not start without jackd running, and there is no "dummy" option in the audio settings.

I tried setting jackd to:
```
-P 70 -t 2000 -s -d dummy -r 44100 -p 256 -n 2 -X raw
```

But I get:
```
root@zynthian:~# systemctl status zynthian.service -l 
● zynthian.service - Zynthian User Interface
   Loaded: loaded (/etc/systemd/system/zynthian.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2018-04-09 11:24:20 UTC; 29s ago
 Main PID: 451 (startx)
   CGroup: /system.slice/zynthian.service
           ├─451 /bin/sh /usr/bin/startx ./zynthian.sh -- :0 vt3
           ├─478 xinit ./zynthian.sh -- /usr/bin/X :0 vt3 -auth /tmp/serverauth.8J3FhTjDMN
           ├─479 /usr/lib/xorg/Xorg :0 vt3 -auth /tmp/serverauth.8J3FhTjDMN
           ├─499 /bin/bash ./zynthian.sh
           └─717 /usr/bin/python3 ./zynthian_gui.py

Apr 09 11:24:47 zynthian startx[451]: ALSA: Cannot open PCM device alsa_pcm for playback. Falling back to capture-only mode
Apr 09 11:24:47 zynthian startx[451]: JackTemporaryException : now quits...
Apr 09 11:24:47 zynthian startx[451]: Cannot initialize driver
Apr 09 11:24:47 zynthian startx[451]: JackServer::Open failed with -1
Apr 09 11:24:47 zynthian startx[451]: Failed to open server
Apr 09 11:24:48 zynthian startx[451]: Cannot connect to server socket err = No such file or directory
Apr 09 11:24:48 zynthian startx[451]: Cannot connect to server request channel
Apr 09 11:24:49 zynthian startx[451]: ERROR:zynautoconnect.zynthian_autoconnect:ERROR Autoconnecting: 'NoneType' object has no att
Apr 09 11:24:49 zynthian startx[451]: Cannot connect to server socket err = No such file or directory
Apr 09 11:24:49 zynthian startx[451]: Cannot connect to server request channel
```

Using net I can get a GUI running, but can't start stuff like pianoteq:
```
jackd -r -p 8 -d net -C 2 -P 0
```

But then I get:
```
Cannot write socket fd = 12 err = Broken pipe
CheckRes error
Could not write notification
ClientNotify fails name = Pianoteq61 notification = 18 val1 = 1 val2 = 0
Cannot write socket fd = 8 err = Broken pipe
CheckRes error

```

jofemodo, 2018-04-09:
I never tried to configure a "dummy" audio device with jackd. Let me take a deeper look ...

jofemodo, 2018-06-06:
Audio dummy device is now listed in the webconf tool. I've tested and it works like a charm, so ... we can write some tests now ;-)