zynthian/zynthian-ui #34: Midi record mode

mheidt, 2018-05-05:
Analog to Audio Recording we should be able to record the midi input.

https://discourse.zynthian.org/t/notes-hanging-zynthian-misses-note-off-solved/1291/86

mheidt, 2018-05-06:
@jofemodo 
how do I record midi data?
arecordmidi -p 14 test2.mid didn't work.
a2jmidid should run...i let him play test.mid first...

jofemodo, 2018-05-08:
Alsa Port 14 is not connected to the zyncoder output by default, so "arecordmidi -p 14 test2.mid" wont work without creating this connection. Also, a2jmidid is not running by default, so yo need to launch a2jmidid before ...

Anyway, i don't like to use ALSA tools. Jitter will be greater and when Zynthian is loaded, we could obtain poor MIDI recordings. We should try to use pure Jack tools. Currently these 2 tools are installed in zynthian images:

+ jack-smf-recorder
+ jack-smf-player

The simpler option for recording MIDI is taking the output from zyncoder, after the MIDI router/filter. Obviously, when replaying we should bypass the MIDI filter too. That is slighty more complex with the current layout, although not too much. 

I'm planning some "big changes" in the way zyncoder works, that will simplify bypassing the filter and many other things ;-)

Regards,

mheidt, 2018-05-10:
So we better wait for the "big changes"?

mheidt, 2018-05-10:
How do you like the idea, that we assign a midi file per layer which can be started and stopped by two keys on that channel.
Guess, we need to modify the file: midi channel and cool would be the BPM via zynthian-ui.

mheidt, 2018-08-03:
How do we proceed here now that the big changes are there?

mheidt, 2018-08-03:
Or is this a duplicate to #35?

jofemodo, 2018-08-09:
Hi @mheidt !

I've been improving the Audio Recorder tool. Now you can "list" and "play" the captured tracks.
It's far from perfect, but useful. The new "zynthian_gui_audio_recorder" class would be a good and simple model for the MIDI Recorder tool. If you want to try it ... if not, i will do it ASAP.

As i tell you, this 2 tools are "fast & dirty" tools. I hope the upcoming "u-DAW" features will replace both tools very soon ;-)

jofemodo, 2018-08-30:
Hi @mheidt!

This feature is implemented in c473ab7ebbe901c2d98b0a9d43feab5abfd9ab98.

closed 2018-08-30 by jofemodo