zynthian/zynthian-ui #35: MIDI Looper/Sequencer

jofemodo, 2018-05-17:
Implement a MIDI looper/sequencer with capability of loading & saving mid files.
The free software "midish" is the best suited option for this:

https://github.com/ratchov/midish

jofemodo, 2019-06-06:
Alexandre Ratchov send me an update by email:

> I don't know if this is still of interest for you, but recently I've
> added initial support for looping. Basically, the current selection
> can be played or recorded in a loop and midish would properly merge in
> real-time conflicting events from the previous periods.
> 
> That's not much for now, but as we've the low-level looping bits,
> adding more sophisticated semantics around it would be much easier
> now.

riban-bw, 2022-11-08:
This is implemented in the MIDI player and step sequencer.

closed 2022-11-08 by riban-bw