zynthian/zynthian-ui #37: SooperLooper full integration

jofemodo, 2018-05-17:
Create a true SooperLooper engine, using the SL server OSC API.

mheidt, 2019-07-15:
Something to consider:

 I added an instance via Detailed view to the instrument.
What I find disturbing is that the Play has to be ON as well, when you try to record.
Mapping the controls to pads worked, when you define them as CC toggle.
There are 2 features I miss, @jofemodo

Play sync. I gonna have one SooperLooper per midi channel especially because the loops will have different lengths. In the UI there is this sync switch the let’s all SP instances start synchronized.
Changing the BPM
And we have an architectural issue.
I mapped my Alesis V25 being MidiChannel11.
I was thinking of recording the instruments on that channel and move them away to a different channel when done…using single channel mode btw…
BUG: after moving, the PlayToggle button still works in the channel overview. But when I go into the channel, it seems to remember, that it isn’t channel 11 anymore and the toggle doesn’t work anymore.
When going back to channel 11, the SP loop is reset as well.

Actually I want my keyboard controlling the toggle after those channel moves.
I think an absolute midi learn that takes the channel into account is not feasible.
But I can imagine a different behaviour for the master channel.
So assuming I programmed my toggles to send the cc as master midi channel, the learned midi commmd should survive the midi channel move.

What do you thing?

mheidt, 2019-07-25:
I had a look into the OSC interface.
I think we should have one "master engine" where you control the generic setup like BPM and where you can define the loops upfront by entering the beats or seconds...I prefer beats. Osc needs time, if I read correctly.
In the control view of the channel there are the pages where you can assign one of those loops and other loop related parameters/actions.
The 0 entry of the loop selector means "New loop". We need to switch to the created loop id, when we end the loop.

jofemodo, 2019-08-01:
Hi @mheidt !

> BUG: after moving, the PlayToggle button still works in the channel overview. But when I go into the channel, it seems to remember, that it isn’t channel 11 anymore and the toggle doesn’t work anymore. When going back to channel 11, the SP loop is reset as well.

This is the normal behaviour of single channel mode. ALL MIDI events (except master channel)) are routed to the active channel, so no way of using the current Sooper Looper implementation in such a way. You should use normal mode (multitimbric) and change the channel in your keyboard.

jofemodo, 2019-08-01:
> Play sync. I gonna have one SooperLooper per midi channel especially because the loops will have different lengths. In the UI there is this sync switch the let’s all SP instances start synchronized.
Changing the BPM

Regarding this, the current implementation (LV2) doesn't allow to sync multiple instances nor change the BPM. We have to implement the full engine, with OSC control for that.

jofemodo, 2022-10-28:
It's implemented by @riban-bw !!

closed 2022-10-28 by jofemodo