zynthian/zynthian-ui #39: Sound stops with program change (with midi rule: no PG)

Hampelschwein, 2018-06-07:
When playing Linux Sampler and sending a program change from my midi keyboard, the sound would stop even though no progam change actually happened.
I am using this Midi filter rule: IGNORE CH#0,1,2,3,4,5,6,7,8,9,10,11,12,13,14 PG

When I go to the main menue afterwards, remove all layers and try to get back to the sound Zynthian crashes in the midi channel selection screen.

`INFO:root:Starting Engine LinuxSampler
INFO:root:Connecting with LinuxSampler Server…
ERROR:root:FAILED lscp_send_single(CREATE AUDIO_OUTPUT_DEVICE JACK ACTIVE=‘true’ CHANNELS=‘2’ NAME=‘LinuxSampler’ SAMPLERATE=‘44100’
): timed out
ERROR:root:FAILED lscp_send_single(SET AUDIO_OUTPUT_CHANNEL_PARAMETER None 0 NAME=‘Channel 1’
): timed out
ERROR:root:0 (ERR)
DEBUG:root:ZCTRL ‘Channel’: 0 (0 -> 16), None, None
INFO:root:Short Switch 3
WARNING:root:zynthian_gui.zyncoder_read() => 0 (ERR)`

I can select other sound engines than Linux sampler still, but there is no sound coming out.

This is how it looks, when I was in another engine before and then went to linux sampler:
`INFO:root:Getting Preset List for Dual
DEBUG:root:PRESET LIST =>
[(’/zynthian/zynthian-ui/data/zynbanks/Dual/0001-Layered1.xiz’, [12, 0, 0], ‘Layered1’, ‘xiz’), (’/zynthian/zynthian-ui/data/zynbanks/Dual/0002-Layered2.xiz’, [12, 0, 1], ‘Layered2’, ‘xiz’), (’/zynthian/zynthian-ui/data/zynbanks/Dual/0005-Organ and Saw.xiz’, [12, 0, 4], ‘Organ and Saw’, ‘xiz’), (’/zynthian/zynthian-ui/data/zynbanks/Dual/0007-Organ Choir Pad1.xiz’, [12, 0, 6], ‘Organ Choir Pad1’, ‘xiz’), (’/zynthian/zynthian-ui/data/zynbanks/Dual/0008-Organ Choir Pad2.xiz’, [12, 0, 7], ‘Organ Choir Pad2’, ‘xiz’), (’/zynthian/zynthian-ui/data/zynbanks/Dual/0033-Rhodes Strings1.xiz’, [12, 0, 32], ‘Rhodes Strings1’, ‘xiz’), (’/zynthian/zynthian-ui/data/zynbanks/Dual/0034-Rhodes Strings2.xiz’, [12, 0, 33], ‘Rhodes Strings2’, ‘xiz’), (’/zynthian/zynthian-ui/data/zynbanks/Dual/0035-Rhodes Strings3.xiz’, [12, 0, 34], ‘Rhodes Strings3’, ‘xiz’), (’/zynthian/zynthian-ui/data/zynbanks/Dual/0036-Rhodes Strings4.xiz’, [12, 0, 35], ‘Rhodes Strings4’, ‘xiz’), (’/zynthian/zynthian-ui/data/zynbanks/Dual/0065-Dream of the Saw.xiz’, [12, 0, 64], ‘Dream of the Saw’, ‘xiz’)]
DEBUG:root:ZCTRL ‘Preset’: 4 (0 -> 10), None, None
INFO:root:Short Switch 1
DEBUG:root:BACK TO SCREEN => bank
DEBUG:root:BANK INDEX => 12
DEBUG:root:BANK LIST =>
[(’/zynthian/zynthian-ui/my-data/zynbanks/XLZ’, 0, ‘MY/XLZ’, ‘MY’), (’/zynthian/zynthian-ui/my-data/zynbanks/XMZ’, 1, ‘MY/XMZ’, ‘MY’), (’/zynthian/zynthian-ui/my-data/zynbanks/XSZ’, 2, ‘MY/XSZ’, ‘MY’), (’/zynthian/zynthian-ui/data/zynbanks/Arpeggios’, 3, ‘Arpeggios’, ‘’), (’/zynthian/zynthian-ui/data/zynbanks/Bass’, 4, ‘Bass’, '’), (’/zynthian/zynthian-ui/data/zynbanks/Brass’, 5, ‘Brass’, ‘’), (’/zynthian/zynthian-ui/data/zynbanks/Choir and Voice’, 6, ‘Choir and Voice’, '’), (’/zynthian/zynthian-ui/data/zynbanks/Collection’, 7, ‘Collection’, ‘’), (’/zynthian/zynthian-ui/data/zynbanks/Companion’, 8, ‘Companion’, '’), (’/zynthian/zynthian-ui/data/zynbanks/Cormi_Noise’, 9, ‘Cormi Noise’, ‘’), (’/zynthian/zynthian-ui/data/zynbanks/Cormi_Sound’, 10, ‘Cormi Sound’, '’), (’/zynthian/zynthian-ui/data/zynbanks/Drums’, 11, ‘Drums’, ‘’), (’/zynthian/zynthian-ui/data/zynbanks/Dual’, 12, ‘Dual’, '’), (’/zynthian/zynthian-ui/data/zynbanks/Fantasy’, 13, ‘Fantasy’, ‘’), (’/zynthian/zynthian-ui/data/zynbanks/Guitar’, 14, ‘Guitar’, '’), (’/zynthian/zynthian-ui/data/zynbanks/Laba170bank’, 15, ‘Laba170bank’, ‘’), (’/zynthian/zynthian-ui/data/zynbanks/Misc’, 16, ‘Misc’, '’), (’/zynthian/zynthian-ui/data/zynbanks/Noises’, 17, ‘Noises’, ‘’), (’/zynthian/zynthian-ui/data/zynbanks/Organ’, 18, ‘Organ’, '’), (’/zynthian/zynthian-ui/data/zynbanks/Pads’, 19, ‘Pads’, ‘’), (’/zynthian/zynthian-ui/data/zynbanks/Plucked’, 20, ‘Plucked’, '’), (’/zynthian/zynthian-ui/data/zynbanks/Reed and Wind’, 21, ‘Reed and Wind’, ‘’), (’/zynthian/zynthian-ui/data/zynbanks/Rhodes’, 22, ‘Rhodes’, '’), (’/zynthian/zynthian-ui/data/zynbanks/Splitted’, 23, ‘Splitted’, ‘’), (’/zynthian/zynthian-ui/data/zynbanks/Strings’, 24, ‘Strings’, '’), (’/zynthian/zynthian-ui/data/zynbanks/Synth’, 25, ‘Synth’, ‘’), (’/zynthian/zynthian-ui/data/zynbanks/SynthPiano’, 26, ‘SynthPiano’, '’), (’/zynthian/zynthian-ui/data/zynbanks/net-wisdom’, 27, ‘net-wisdom’, ‘’), (’/zynthian/zynthian-ui/data/zynbanks/olivers-100’, 28, ‘olivers-100’, '’), (’/zynthian/zynthian-ui/data/zynbanks/olivers-other’, 29, ‘olivers-other’, ‘’), (’/zynthian/zynthian-ui/data/zynbanks/the_mysterious_bank’, 30, ‘the mysterious bank’, '’), (’/zynthian/zynthian-ui/data/zynbanks/the_mysterious_bank_2’, 31, ‘the mysterious bank 2’, ‘’), (’/zynthian/zynthian-ui/data/zynbanks/the_mysterious_bank_3’, 32, ‘the mysterious bank 3’, '’), (’/zynthian/zynthian-ui/data/zynbanks/the_mysterious_bank_4’, 33, ‘the mysterious bank 4’, ‘_’)]
DEBUG:root:ZCTRL ‘Bank’: 12 (0 -> 34), None, None
INFO:root:Short Switch 1
DEBUG:root:BACK TO SCREEN => layer
DEBUG:root:ZCTRL ‘Layer’: 0 (0 -> 5), None, None
INFO:root:Short Switch 3
DEBUG:root:ZCTRL ‘Layer’: 0 (0 -> 4), None, None
INFO:root:Stoping Engine ZynAddSubFX
INFO:root:Short Switch 3
DEBUG:root:ZCTRL ‘Engine’: 0 (0 -> 7), None, None
INFO:root:Short Switch 3
INFO:root:Starting Engine LinuxSampler
INFO:root:Connecting with LinuxSampler Server…
ERROR:root:FAILED lscp_send_single(CREATE AUDIO_OUTPUT_DEVICE JACK ACTIVE=‘true’ CHANNELS=‘2’ NAME=‘LinuxSampler’ SAMPLERATE=‘44100’
): timed out
ERROR:root:FAILED lscp_send_single(SET AUDIO_OUTPUT_CHANNEL_PARAMETER None 0 NAME=‘Channel 1’
): timed out
ERROR:root:FAILED lscp_send_single(SET AUDIO_OUTPUT_CHANNEL_PARAMETER None 1 NAME=‘Channel 2’
): timed out
ERROR:root:FAILED lscp_send_single(SET AUDIO_OUTPUT_CHANNEL_PARAMETER None 0 JACK_BINDINGS=‘system:playback_1’
): timed out
ERROR:root:FAILED lscp_send_single(SET AUDIO_OUTPUT_CHANNEL_PARAMETER None 1 JACK_BINDINGS=‘system:playback_2’
): timed out
ERROR:root:FAILED lscp_send_single(CREATE MIDI_INPUT_DEVICE JACK ACTIVE=‘true’ NAME=‘LinuxSampler’ PORTS=‘1’
): timed out
ERROR:root:FAILED lscp_send_single(SET VOLUME 0.45
): timed out
DEBUG:root:ZCTRL ‘Channel’: 0 (0 -> 16), None, None
INFO:root:Short Switch 3
WARNING:root:zynthian_gui.zyncoder_read() => 0 (ERR)`

jofemodo, 2018-06-25:
Hi @Hampelschwein!

Please, update your zynthian software and test again. I fixed some bug and perhaps it's related with your problem. Test without MIDI rules first.

Regards,

jofemodo, 2018-06-25:
I've tested and yes, there is a bug in the MIDI router that causes engines to crash with certain types of rules (PG and perhaps others too). I'm working to fix it ASAP.

In the other side, the MIDI Program Change works perfectly when no MIDI router rules are defined. I've tested with several engines and i can't see any problem with it.

Regards,

jofemodo, 2018-07-29:
It should be fixed with the latest changes. Please, can confirm?

Hampelschwein, 2018-07-29:
thanks for letting me know! I will check next week, I'm in Italy right now :-)