zynthian/zynthian-ui #43: Master Program Change

mheidt, 2018-07-05:
I couldn't confirm myself yet, but the snapshot change via Midi program change up/down  doesn't seem to work

https://discourse.zynthian.org/t/master-program-change-up-down-feature/2036/8

mheidt, 2018-07-22:
I can confirm it.
@jofemodo 
I would expect code in zynthian_zcmidi.
But I find bank and prg-select only.

Is this feature implemented yet?
Or would you do it differently with the new midi filter anyways?

jofemodo, 2018-07-22:
AFAIK it's implemented:

https://github.com/zynthian/zynthian-ui/blob/02d07d14277ff20284fa2a278c3d8859efb34973/zynthian_gui.py#L485

Anyway, i will test the feature ASAP, i hope tomorrow i will have my summer-lab ready to work ;-)

jofemodo, 2018-07-27:
I've been testing this feature in a zynthian updated to the last changes and found a bug related with the recently added feature "active channel". After fixing it, everything works as expected. Could somebody confirm? 

Thanks,

mheidt, 2018-07-28:
ok, i can make it work, but I can't make it change.

[image]
I rebooted the machine after I changed the webconf settings.
I changed every value and nevertheless the program change is executed

jofemodo, 2018-07-29:
It seems that the problem is the way you have organized your snapshots. Try putting your snapshots inside banks. Anyway, i will try to improve the engine code to support bank-less arrangements.

mheidt, 2018-07-29:
Actually I have...
But I wonder why it reacts on 00 although I user different messages. I will add more log messages tonight.

mheidt, 2018-07-29:
INFO:root:MASTER MIDI MESSAGE: 0xcf5a00
DEBUG:root:PROGRAM CHANGE UP!
DEBUG:root:PROGRAM CHANGE 90

if ev==zynthian_gui_config.master_midi_bank_change_up:
guess, it needs to be elif

I checked into master

jofemodo, 2018-07-30:
OK! Does this change fix the problem?

mheidt, 2018-07-30:
In my case. Claude is going to test it hopefully as well. At least he is aware, as he gave me a like.

jofemodo, 2018-08-12:
As it's working for me and for you, i suposse it's solved ;-)

closed 2018-08-12 by jofemodo