zynthian/zynthian-ui #47: Possible Bug: Pianoteq Engine - Order of Execution

Jonnie-Cache, 2018-08-02:
I've been trying to track down what I think is still a bug in the creation of the menu item for Pianoteq. It's tough because I still don't have a working ability to step through the code to see what happens when. I've just used logging to see if I can figure it out, but I feel like I'm stuck or at least need some direction. Hopefully someone can help point me in the right direction.

I believe there is a problem in the order of execution. In zynthian_gui_engine.py, line 61 is looking for the PIANOTEQ envs. But, I think that zynthian_engine_pianoteq.py lines 136 and 137 are set up before they are actually initialized by get_pianoteq_binary_info():.

Further, I think that line 60 in zynthian_gui_engine.py should be calling get_pianoteq_binary_info(): instead of check_pianoteq_binary().

If I'm understanding the code correctly, what happens is that zynthian_engine_pianoteq.py lines 136 and 137 are executed. But, PIANOTEQ_STAGE is initialized with a 1 even when it should be 0. I think that's because get_pianoteq_binary_info(): hasn't been called yet. So, the result is that the menu will always say "STAGE" on it when it shouldn't.

If I'm correct, I think I might have a fix for this. But hoping that someone else can at least validate that what I'm seeing is correct before I commit and submit a pull request.

jofemodo, 2018-08-12:
Hi @Jonnie-Cache,

The zynthian engine takes the Pianoteq binary info from the environment variables.It doesn't call to the "get_pianoteq_binary_info" because the Pianoteq binary takes a lot of time to load and it will "double" the engine's starting time, that it's quite long now.

The Pianoteq's environment variables configuration is done by the webconf tool when installing the binary (it calls the "get_pianoteq_binary_info()!!), and saved in the "/zynthian/config/zynthian_envars.sh" file. After modifying this file, a reboot is highly recommended ;-)

Jonnie-Cache, 2018-08-13:
Ah. That makes sense. Was unaware of zynthian_envars.sh. I'll have to take a look at that.

I think that what I'll need to look at, then, is the script for the webconf tool to permit the Pro version to be installed. I had just been doing the installation manually.

jofemodo, 2018-09-05:
Hi @Jonnie-Cache!
Did you implemented the Pro version installation procedure in the webconf tool?

Jonnie-Cache, 2018-09-05:
I'm sorry, I didn't have time for a bit and then my development machine died.

closed 2019-02-19 by jofemodo