zynthian/zynthian-ui #49: New No/Yes button easy to press NO on lower functions (Reboot etc)

wyleu, 2018-08-17:
The location of the No button makes it very eay for this to be selected when you press functions like reboot.
It looks like a failure because effectively you select No unintentionally.

Relocate No Button?

mheidt, 2018-08-17:
The NO is left, isn't it?
The right lower button is the Select button, which makes it YES Freitag,

jofemodo, 2018-08-17:
I agree with @mheidt. The "NO" button is perfectly OK ;-)
@wyleu, could you be more explicit about your use case? Are you using the touch interface?

wyleu, 2018-08-17:
Using it via mouse on hdmi screen. It's quite difficult to reboot this
way.  It wasn't until I worked out what was going on I thought it wasn't
rebooting. Ignore if it's just another wyleu whinge

jofemodo, 2018-08-17:
@mheidt, perhaps we could add an option for disabling the "confirmation dialog" ?
It should be located in the "User Interface" tab ...

jofemodo, 2018-08-31:
Please @wyelu, explain the problem a little bit more in depth. I would like to understand the issue too ... ;-)

wyleu, 2018-09-01:
Don't worry. It's a familiarity issue on my part.
False alarm

closed 2018-10-01 by jofemodo