zynthian/zynthian-ui #53: Problem using jalv on desktop

jfrey-xx, 2018-10-04:
Hello,

I am using Zynthian on Ubuntu 16.04. I pulled the last changes implementing jalv (which I compiled from zynthian branch), but I came up with a strange error upon adding a new layer `ERROR:root:Can't exec engine commandoo: \info_controls => an integer is required (got type NoneType)`. The stack trace, once enabled, gives more details:

```
Traceback (most recent call last):
  File "/home/path/to/zynthian/zynthian-ui/zyngine/zynthian_engine_jalv.py", line 205, in proc_cmd
    self.proc.sendline(cmd)
  File "/usr/lib/python3/dist-packages/pexpect/pty_spawn.py", line 524, in sendline
    n = self.send(s)
  File "/usr/lib/python3/dist-packages/pexpect/pty_spawn.py", line 509, in send
    time.sleep(self.delaybeforesend)
TypeError: an integer is required (got type NoneType)
```
There is an easy fix, either disabling line 181 in `zynthian_engine_jalv.py`, i.e. `			self.proc.delaybeforesend = None` to keep default value, or setting it explicitly to 0 if the goal is to disable that. I don't know what are the side effects, if any, on my system everything seems to be fine with both solution, and I can finally enjoy my preferred synths with a simple interface that I can pilot with my midi keyboard with some more OSC-based hacks.

If this is of interest, I can give more details about my system and the versions of the various libs/soft installed.

Thanks a lot for this piece of software!

jofemodo, 2018-10-04:
Hi @jfrey-xx !

The problem seems related to your version of "pexpect". In Raspbian it works OK.
Anyway, i've changed the line to:

`self.proc.delaybeforesend = 0`

In such a way, it will work in both platforms ;-)

Thanks for your feedback!!

closed 2018-10-04 by jofemodo

jfrey-xx, 2018-10-05:
It is not relevent anymore, just for closure. pexpect is version 4.0.1-1 on my system, from ubuntu 16.04 repos.

Thanks for the fix!

commit 7608e6ca (2018-10-04): Fix #53: Solve problem using Jalv on desktop (Ubuntu 16.04)
Pexpect fails when setting "self.proc.delaybeforesend = None". Better set to 0 explicitly.