zynthian/zynthian-ui #55: Improve Layer Audio-Routing

jofemodo, 2018-10-08:
Allowing to connect engine's audio-outputs to any available audio-input, including "system", "mod-monitor", etc.

jofemodo, 2018-10-08:
It's almost ready, but it would be nice to separate "Filter Plugins" from "Generator Plugins" and having 2 "New" entries in the Layer Screen:

+ New Synth Layer
+ New Effect Layer

Or something like that ... ;-)

mheidt, 2018-10-08:
Do you know, how I see the difference? The API didn't show much. At least where I looked. Or do I have to look in a certain file?

jofemodo, 2018-10-09:
Take a look at this ;-)

```
#!/usr/bin/env python3

import lilv

world = lilv.World()
world.ns.ev = lilv.Namespace(world, 'http://lv2plug.in/ns/ext/event#')

def get_plugin_type(pl):
	global world

	n_audio_in  = pl.get_num_ports_of_class(world.ns.lv2.InputPort,  world.ns.lv2.AudioPort)
	n_audio_out = pl.get_num_ports_of_class(world.ns.lv2.OutputPort, world.ns.lv2.AudioPort)
	n_midi_in  = pl.get_num_ports_of_class(world.ns.lv2.InputPort,  world.ns.ev.EventPort)
	n_midi_out = pl.get_num_ports_of_class(world.ns.lv2.OutputPort, world.ns.ev.EventPort)

	if n_audio_in>0 and n_audio_out>0 and n_midi_in==n_midi_out==0:
		return "Audio Effect"
	if n_audio_out>0 and n_midi_in==n_midi_out==0:
		return "Audio Generator"
	if n_midi_in>0 and n_midi_out>0 and n_audio_in==n_audio_out==0:
		return "MIDI Tool"
	if n_audio_out>0 and n_midi_in>0:
		return "MIDI Synth"
	return "Unknown"

world.load_all()
for p in world.get_all_plugins():
   print("{} ({}) => {}".format(p.get_name(),get_plugin_type(p),p.get_uri()))
```

jofemodo, 2018-10-09:
At first, i tried to use the "plugin.get_class" function, but with no result. It always return "LV2Plugin"...
So, i dived deeper and created this little "get_plugin_type()" function ... ;-)

Enjoy!

jofemodo, 2018-10-09:
Slighty better:

```
#!/usr/bin/env python3

import lilv

world = lilv.World()
world.ns.ev = lilv.Namespace(world, 'http://lv2plug.in/ns/ext/event#')

def get_plugin_type(pl):
	global world

	n_audio_in  = pl.get_num_ports_of_class(world.ns.lv2.InputPort,  world.ns.lv2.AudioPort)
	n_audio_out = pl.get_num_ports_of_class(world.ns.lv2.OutputPort, world.ns.lv2.AudioPort)
	n_midi_in  = pl.get_num_ports_of_class(world.ns.lv2.InputPort,  world.ns.ev.EventPort)
	n_midi_out = pl.get_num_ports_of_class(world.ns.lv2.OutputPort, world.ns.ev.EventPort)

	if n_audio_out>0 and n_midi_in==n_midi_out==0:
		if n_audio_in>0:
			return "Audio Effect"
		else:
			return "Audio Generator"

	if n_midi_in>0 and n_midi_out>0 and n_audio_in==n_audio_out==0:
		return "MIDI Tool"

	if n_audio_out>0 and n_midi_in>0:
		return "MIDI Synth"

	return "Unknown"

world.load_all()
for p in world.get_all_plugins():
   print("{} ({}) => {}".format(p.get_name(),get_plugin_type(p),p.get_uri()))
```

mheidt, 2018-10-09:
The current webconf code is buggy.
The file contains "INSTALLED" (your change), but webconf code uses "ENABLED".
Do you have some uncommited changes....before I start...

jofemodo, 2018-10-09:
I changed everything to "enabled". If there is still some "installed", it should be changed ...

mheidt, 2018-10-09:
ok! my file is wrong then :)

mheidt, 2018-10-09:
How do you need the result? A type in the existing file(s) or 2 different config files?

jofemodo, 2018-10-09:
A type in the existing file is perfect for me if you like ;-)

mheidt, 2018-10-09:
There are Unknown types:

Unknown MIDI CC Map http://gareus.org/oss/lv2/midifilter#mapcc
Unknown MIDI CC to Note http://gareus.org/oss/lv2/midifilter#cctonote
Unknown MIDI Channel Filter http://gareus.org/oss/lv2/midifilter#channelfilter
Unknown MIDI Channel Map http://gareus.org/oss/lv2/midifilter#channelmap
Unknown MIDI Channel Unisono http://gareus.org/oss/lv2/midifilter#mididup
Unknown MIDI Chord http://gareus.org/oss/lv2/midifilter#midichord
Unknown MIDI Chromatic Transpose http://gareus.org/oss/lv2/midifilter#miditranspose
Unknown MIDI Clock Generator http://gareus.org/oss/lv2/mclk
Unknown MIDI Delayline http://gareus.org/oss/lv2/midifilter#mididelay
Unknown MIDI Duplicate Blocker http://gareus.org/oss/lv2/midifilter#nodup
Unknown MIDI Enforce Scale http://gareus.org/oss/lv2/midifilter#enforcescale
Unknown MIDI Event Filter http://gareus.org/oss/lv2/midifilter#eventblocker
Unknown MIDI Key-Range Filter http://gareus.org/oss/lv2/midifilter#keyrange
Unknown MIDI Keysplit http://gareus.org/oss/lv2/midifilter#keysplit
Unknown MIDI Monophonic Legato http://gareus.org/oss/lv2/midifilter#monolegato
Unknown MIDI N-Tap Delay http://gareus.org/oss/lv2/midifilter#ntapdelay
Unknown MIDI Note Toggle http://gareus.org/oss/lv2/midifilter#notetoggle
Unknown MIDI Note Transpose http://gareus.org/oss/lv2/midifilter#mapkeyscale
Unknown MIDI Note to CC http://gareus.org/oss/lv2/midifilter#notetocc
Unknown MIDI Note/Channel Map http://gareus.org/oss/lv2/midifilter#mapkeychannel
Unknown MIDI Quantization http://gareus.org/oss/lv2/midifilter#quantize
Unknown MIDI Remove Active Sensing http://gareus.org/oss/lv2/midifilter#noactivesensing
Unknown MIDI Scale CC Value http://gareus.org/oss/lv2/midifilter#scalecc
Unknown MIDI Simple Channel Filter http://gareus.org/oss/lv2/midifilter#onechannelfilter
Unknown MIDI Sostenuto http://gareus.org/oss/lv2/midifilter#sostenuto
Unknown MIDI Step Sequencer16x16 http://gareus.org/oss/lv2/stepseq#s16n16
Unknown MIDI Step Sequencer4x4 http://gareus.org/oss/lv2/stepseq#s4n4
Unknown MIDI Step Sequencer4x8 http://gareus.org/oss/lv2/stepseq#s4n8
Unknown MIDI Step Sequencer8x8 http://gareus.org/oss/lv2/stepseq#s8n8
Unknown MIDI Strum http://gareus.org/oss/lv2/midifilter#midistrum
Unknown MIDI Thru http://gareus.org/oss/lv2/midifilter#passthru
Unknown MIDI Velocity Adjust http://gareus.org/oss/lv2/midifilter#velocityscale
Unknown MIDI Velocity Randomization http://gareus.org/oss/lv2/midifilter#randvelocity
Unknown MIDI Velocity-Range Filter

mheidt, 2018-10-09:
You need to change your part now.
config/jalv_plugins.json changed. I committed into master.
Don't drop the ball :)

jofemodo, 2018-10-10:
Done and committed to master ;-)

closed 2018-10-10 by jofemodo