zynthian/zynthian-ui #57: my data

spurkopf, 2018-10-30:
That really made me headache. I could not start webconf with a new self-soldered zynthian box.
Till I found this:

``Traceback (most recent call last):
File "/usr/local/lib/python3.4/dist-packages/tornado/web.py", line 1323, in _execute result = self.prepare()
File "/zynthian/zynthian-webconf/lib/zynthian_config_handler.py", line 46, in prepare
    zynconf.load_midi_config()
  File "/zynthian/zynthian-ui/zynconf/zynthian_config.py", line 94, in load_midi_config
    return load_config(set_env, fpath)
  File "/zynthian/zynthian-ui/zynconf/zynthian_config.py", line 56, in load_config
    with open(fpath) as f:
FileNotFoundError: [Errno 2] No such file or directory: '/zynthian/zynthian-ui/my-data/midi-profiles/default.sh'
ERROR:tornado.access:500 GET / (192.168.1.101) 44.13ms
INFO:tornado.access:200 GET /favicon.ico (192.168.1.101) 16.82ms
^CTraceback (most recent call last):
  File "./zynthian_webconf.py", line 147, in <module>
    tornado.ioloop.IOLoop.current().start()``

/midi-profiles/default.sh 
is not in /zynthian/zynthian-ui/my-data/
it´s in /zynthian/zynthian-my-data/

jofemodo, 2018-10-30:
It seems a very old problem that is already solved ...
What SD image are you using? Have you updated the image to the latest version?

Thanks,

spurkopf, 2018-10-30:
I used _zynthian_gorgona_omega_rbpi3_kitv2-2018-04-03_
and yes the device has been updated.

jofemodo, 2018-10-31:
Strange. Webconf works perfectly OK out-the-box. Probably there was some error on the update process. First, try updating again. If it doesn't work, please, try re-burning the SD and update from the command line.

Regards,

spurkopf, 2018-10-31:
That's what i posted before i posted the issue:
copied zynthian_gorgona_omega_rbpi3_kitv2-2018-04-03 with etcher on a new 64 GB SD card.
ran updates on RJ45 a few times.
- no webconf
Then 2 days later, on a different new SD card - again copied the image on SD. Immediately tried to gain access to webconf - again not possible. Then again running updates on RJ45 a few times.  
- no webconf
As workaround I copied /midi-profiles/default.sh to /zynthian/zynthian-my-data/
so i have a working system. Having access even over mobile-phone.

Yes I can start again and then try to update from the command line, but I´m pretty sure
there is a bug in the code.

spurkopf, 2018-11-03:
even more strange:
again copied the image an SD-Card - and now updated from the command line. This time worked as expected, I could start webconf. I had tried it also before the update and it worked here too.
... sorry for making noise.

closed 2018-11-03 by spurkopf