zynthian/zynthian-ui #59: Panic Switch behaviour.

wyleu, 2018-12-13:
The Panic switch wants to:

    on short press
    (Press>Release within short press window)
    Send Notes off.

    Once you get passed the short press time and hit the long press
    (Press >>>> outside of short press window)
    Send Sounds (all notes all channels) off.

Otherwise you press and hold it and it’s waiting for the release . . . .and you kind of deadlock.

mheidt, 2019-02-15:
we should make it configurable.
I like to have a Zynthian-UI reboot on a long press f.e.

jofemodo, 2019-02-19:
OK! This is the list of assignable actions i have:

+ POWER_OFF
+ REBOOT
+ ALL_SOUNDS_OFF
+ ALL_NOTES_OFF
+ ALL_OFF => ALL_NOTES_OFF + ALL_SOUNDS_OFF

Do you have some suggestion to add to the list?

mheidt, 2019-02-19:
UI_RESET, which only restarts the service

Do you want to assign numbers (n²) to work with a bit-mask for the  ALL_OFF?

I could write something in the Wiring page.
The variable PANIC_SWITCH_BEHAVIOUR would contain a number of that ENUM :p  

POWER_OFF = 1
REBOOT = 2
ALL_SOUNDS_OFF = 4
ALL_NOTES_OFF = 8
UI_RESET = 16

And for ALL_OFF i will assign 12

jofemodo, 2019-02-19:
+ OK to use n2 logic.
+ We would like to configure the 4 extra switches for triggering one of this actions:
    + UI command (using the list above with n2 logic)
    + Sends ON/OFF MIDI event (Sustain pedal, Mute pedal, Portamento, etc.) to the active channel

Also, it would be good to take into account that in the future we could have analog inputs that could be assigned to:

     + ALSA Volume
     + MIDI CC
     + MIDI CP (Channel Pressure)
     + CV (Gate would be an extra switch configured to send MIDI Note On/Off)
     + Etc.

Regards,

mheidt, 2019-02-19:
How do we name the switches?
CUSTOM_SWITCH_A to D?
What happens, if you want analog ones?
You need 3 pins for it? Hence it might be possible to have 4 binary ones or 1 analog and 1 binary?

mheidt, 2019-02-19:
I have checked something into webconf.

jofemodo, 2019-02-20:
OK! But wait a little bit for the UI implementation. Please, move the last commit on webconf to a separated branch. 

First of all, i would prefer to numerate the custom switches using numbers: CUSTOM_SWITCH_01, CUSTOM_SWITCH_02, etc.

Also, we need an extra "enum" value for "MIDI_MESSAGE", and when selected, extra info should be requested: 

- Event Type
- Event Number
- ON value
- OFF value

For instance, you would like to define a Sustain Pedal switch like that:

- Event Type: CC
- Event Number: 64
- ON value: 127
- OFF value: 0

OK! ;-DDD 

Perhaps it's too complex and most of potential users only want the possibility of adding a "Sustain Pedal" or a "Mute Pedal", so we could make a partial implementation requesting only the CC number, that is what currently implements the UI. If we have users asking for the full functionality, we will add it later. What do you think?

Finally, no need for "ANALOG" enum value. Analog inputs will be defined separately. We will have "ANALOG_INPUT_01" and so on ... Don't do nothing about it by now, only think about this possibility from a "code structure" point of view ...

Thanks!

closed 2019-02-28 by jofemodo