zynthian/zynthian-ui #65: zynaddsubfx and Frames per period

mheidt, 2019-02-12:
I am using Audioinjector Ultra and  changed sample rate from 48000 down to 32000 because pianoteq wasn't performing well.
But at 32000 Pianoteq suggests to change the frames per period (-p) from 256 to 192 or 288.
When I change that setting, zynaddsubfx doesn't start up because it needs the same frames per period.
I guess, it is hardcoded somewhere?
I ended up using a sample rate of 24000 and 256 again.
Hence a low prio bug

jofemodo, 2019-02-12:
Pianoteq has its own internal rate that is configured by default in Zynthian to 22050.
Perhaps your problems with Pianoteq are because of the internal sample rate is not well suited for an external sample rate of 48000. I will try changing the Pianoteq internal sample rate to 24000 ...

mheidt, 2019-02-12:
Changing the sample rate wasn't really the issue. When I change it for jacks, it is changed in Pianoteq as well.
But when you check the performance screen in Pianoteq, it says that the framesize is suboptimal.
When change it to the suggested values, Pianoteq works fine and takes all jackd values.
But ZynAddsubfx doesn't work anymore if the size is 192 f.e.

jofemodo, 2019-02-13:
From the jackd man page:

> **-p, --period int**
> 
> Specify the number of frames between JACK process() calls. This value must be a power of 2, and the default is 1024. If you need low latency, set -p as low as you can go without seeing xruns. A larger period size yields higher latency, but makes xruns less likely. The JACK capture latency in seconds is --period divided by --rate.

I've been researching a little bit and currently you can specify a value not being a power of 2, but most of software using jackd won't work with it.

jofemodo, 2019-02-13:
Take a look to this:

https://www.linuxmusicians.com/viewtopic.php?t=10707

closed 2019-02-13 by jofemodo

mheidt, 2019-02-13:
Ok. I close it

mheidt, 2019-02-13:
Or you :)