zynthian/zynthian-ui #66: Single Channel Mode and Sustain

mheidt, 2019-02-19:
I am running in Single Channel Mode and have a Pianoteq layer on channel 1 and a zynaddsubfx on channel2.

When I hold the sustain pedal and switch from Pianoteq to zynaddsubfx, the sound is held like expected.
But when I release the sustain pedal, the sound doesn't mute, which I like to see.

When I am on channel 2 on the zynaddsubfx holding the sustain pedal and switch back to channel 1, the sound mutes, which i wouldn't suspect.

jofemodo, 2019-02-19:
It's not easy to solve. I will explain the current status:

Currently, when changing the active channel, Zynthian send "All Notes Off" to the current channel before changing to the new. It's needed because currently Zynthian router doesn't know what notes are ON on every channel, so i implemented this simple (but not too good!) way of avoiding the problem of hanging notes. By the contrary, sustain pedal is not turned to off when changing the active channel, so if it's hold when changing the active channel, the sustain pedal will remain ON on the first channel although you release it later, from the second channel.

Regarding your example, Pianoteq ignores "All Notes Off" event, so you are describing accurately the expected behaviour.

For solving the problem, first we need that the Zynthian MIDI router track the active notes for every channel, including the sustain pedal status and perhaps others. So, when a note-off or sustain-off is received in the new active channel, it could check if it should be redirected to the last active channel.

Perhaps i could implement a quick partial solution:
+ On UI: Remove the "All_Notes_Off" event when changing the active channel
+ On MIDI Router: Keep track of sustain pedal status and redirect the sustain-off event to the last active channel.

What do you think?

mheidt, 2019-02-19:
what if you defer the ALL_NOTES_OFF, if sustain is ON.
So when Sustain is off and you have a defered event, you sent ALL_NOTES_OFF.
And you send the Sustain OFF to all channels, not only the current one.

jofemodo, 2019-02-19:
OK! I will give it a try ... ;-)

mheidt, 2019-03-18:
You are not trying hard enough ;-)
Another question regarding Single Channel Mode.
I could imagine a Master Channel Mode as well, which behaves like SCM but only listens to the events on Master MIDI channel.
And how can we switch the layers remotely? Maybe via MIDI program change?

jofemodo, 2019-03-18:
Jijiji! It's a quite dense subject ... i need a big window before jumping there!! ;-)

riban-bw, 2019-05-10:
I too feel the note handling whilst switching patches could be improved. The _quick_ solution of tracking sustain and deferring all_note_off sounds good but it would be great if we could consider tracking all notes too so that a held note during a patch change sustained until released. That is certainly the behaviour I would expect. It would probably require a list of current notes for each layer which is added to for each note on and deleted from for each note off. In theory we then would not need the all_notes_off message. This would work for single channel mode (the subject of this request) but it would be good to see if we could make it work for ZS3 too. It is a typical (and IMO expected) behaviour.

mheidt, 2019-05-10:
my gut tells me, that holding all note events in a list and managing that list costs too much latency.
Something that needs to be tested.

mheidt, 2020-02-10:
with #104 this should be solved now

jofemodo, 2020-02-10:
I think it's not solved yet, although i've a plan for solving it ;-)