zynthian/zynthian-ui #71: MIDI Out and Hz Tuning

mheidt, 2019-03-30:
The Hz tuning uses MIDI pitch bend messages.
Those are correctly send to the internal engines.
But this pitchbend offset should not be considered in the MIDI OUT, because cascaded MIDI devices will be detuned.
As user you don't want to keep in mind, whether you cascade them Keyboard-ZynthA-ZynthB.
It should behave the same if you hook them up Keyboard-ZynthB-ZynthA without changing the tuning settings

jofemodo, 2019-03-30:
Hi Markus!
I'm thinking of improving the fine-tuning system for taking profit of the fine-tuning features that already have some of the engines. ZynAddSubFX, FluidSynth, LinuxSampler, Pianoteq and perhaps others, have per-note fine tuning (using the MTS: https://en.wikipedia.org/wiki/MIDI_tuning_standard), what allows to use different temperaments and exotic scales.
This fits very well with the new "not-tuned" pipe inside the MIDI-router ...   ;-)

jofemodo, 2019-03-31:
It's done! Please, test it ...

mheidt, 2019-04-02:
Please don't close it, when it's not tested yet.
Now the Midi-OUT doesn't work at all.
I have rebooted. (ui-git 91d2dc1)
The ZynMidiRouter Main still shows pitchbends.
And the MIDI-OUT doesn't show any message at all.
The M indicator of the receiving Zynthian doesn't show either

jofemodo, 2019-04-02:
Hi @mheidt !

The ZynMidiRouter Main is OK like that. This is for internal use. The new "non-tuned" port is "ZynMidiRouter MIDI-OUT" ;-)

It works, but the default MIDI profile configuration is wrong. Please, select the MIDI_out in the webconf's "MIDI ports" configuration tab.

Regards,

jofemodo, 2019-04-02:
Regarding the different use-cases ... have you thought that somebody would want to use Zynthian as a "Fine-Tuner" for some synthesizer that doesn't support fine-tunning? In such a case, it could be interesting to have the fine-tunning messages in the MIDI-OUT ... ;-)

mheidt, 2019-04-02:
yes, it works. great. The Midi-IN of the receiving doesn't show pitch events..
Can that other usecase be done by changing the midi port wiring in the webconf?

closed 2019-04-02 by mheidt

jofemodo, 2019-04-02:
>Can that other usecase be done by changing the midi port wiring in the webconf?
Yes. That's is the idea. It's really simple to implement ;-)

commit e4b470da (2019-03-31): Fix #71: Keep fine-tuning messages (pitch-bending) inside Zynthian. Use new "midi_out" ZMOP port for MIDI-OUT.
commit ee43b7e8 (2019-03-31): Fix #71: Keep fine-tuning messages (pitch-bending) inside Zynthian. Use new "midi_out" ZMOP port for MIDI-OUT.