zynthian/zynthian-ui #74: UI key binding

riban-bw, 2019-04-06:
Using a (QWERTY) USB keyboard to control the UI isn't easy (possible). Up / down keys don't navigate lists until after tab is used to set focus and I can't find a way to select an option or go back. It would be advantageous to allow control of the UI from keyboard with some shortcut key bindings. This may also assist with accessibility.

wyleu, 2019-04-11:
I've found using a mouse provides good access ability to the gui but adding keybindings would help standardise interface interaction.

jofemodo, 2019-04-16:
It's quite easy to implement with the new CUIA API. I will do, but it's not a priority.

riban-bw, 2019-05-07:
I have had a quick look at this (because I find it handy to use keyboard input when testing) and it looks like keyboard binding should be fairly simple. I will open a PR / branch for this and start implementation.

jofemodo, 2019-05-07:
Perfect! It's super-easy to implement using the CUIA ;-)

riban-bw, 2019-05-07:
I have been going round in circles with this one. I wanted to provide a way to present the encoders on the keyboard, like having a key that represents the ENTER button that when pressed, sends short, bold or long message based on how long the button is pressed. This allows testing of the physical user interface without the encoders but I don't think this is really of much benefit. The code required would not actually test any of the encoder code and it may prove challenging with little benefit.

I also had the idea of using keyboard binding for keys to encoder buttons but using modifiers for bold and long, e.g.

Key | Modifier | Action
-|-|-
ENTER | None | Select
ENTER | Shift | Bold Select
ENTER | Ctrl | Long Select
Backspace | None | Back
Backspace | Shift | Bold Back
etc. | etc. | etc.
Escape | None | Back
L | None | Layer
S | None | Snapshot

I am tempted to implement these directly (rather than using CUIA) because the coding is simple and less overhead. What do you think?

I would map these statically initially with a future option to map via web conf. I would also add binding for other CUIA, e.g. panic, record, stop, etc.

I would quite like to have the up / down keys to select the page of control parameters list to avoid needing to change modes but it should not select list entries anywhere else. (I think that makes more sense.)

wyleu, 2019-05-07:
I would map these statically initially with a future option to map via web conf. I would also add binding for other CUIA, e.g. panic, record, stop, etc.

We would probably produce a grammar that could map quite nicely to midi keys....

I like the idea of messages running along side encoder stuff. IT means we have one simple control environment, and if we get it right it shouldn't really matter where the source is and on what  . . . 
It can go quite fast over qmidinet :-D

riban-bw, 2019-05-07:
Okay - first draft committed and pushed to branch gh74-keybinding. Bindings:

Key|Modifier|Function
-|-|-
Enter|None|SELECT
Enter|Shift|Bold SELECT
Enter|Ctrl|Long SELECT
Backspace|None|BACK
Backspace|Shift|Bold BACK
Backspace|Ctrl|Long BACK
Escape|None|BACK
Escape|Shift|Bold BACK
Escape|Ctrl|Long BACK
L|None|LAYER
L|Shift|Bold LAYER
L|Ctrl|Long LAYER
S|None|SNAPSHOT
S|Shift|Bold SNAPSHOT
S|Ctrl|Long SNAPSHOT
R|None|Start audio Record
R|Shift|Stop audio Record
M|None|Start MIDI Record
M|Shift|Stop MIDI Record
M|Ctrl|Reload MIDI Config
Space|None|All Notes Off
Space|Shift|All Sounds Off
Space|Ctrl|All off
Insert|None|Restart UI
Insert|Shift|Reboot
Insert|Ctrl|Power Off
Up|None|Listbox Up
Down|None|Listbox Down

riban-bw, 2019-05-09:
Branch https://github.com/zynthian/zynthian-ui/tree/gh74-keybinding now committed with what I think is a working implementation. Please review.

jofemodo, 2019-06-20:
It's merged on master now ;-)
Thanks @riban-bw !

closed 2019-06-20 by jofemodo

jofemodo, 2019-06-20:
Added "Left" for "Back-action" & "Right" for "Select-action", so you can navigate using the 4 arrows (3 fingers navigation!).

jofemodo, 2019-06-20:
Somebody should write a Wiki entry for this feature, including the key table ;-)

riban-bw, 2019-06-20:
I wonder who "_somebody_" will be??? I can do something - where within the wiki do you think this lives?

jofemodo, 2019-06-20:
In the UI user's guide, of course:

http://wiki.zynthian.org/index.php/Zynthian_UI_Users_Guide

Perhaps it should be splited in several pages. It's growing too much ...

riban-bw, 2019-06-20:
I have updated that page with a table based on what is in this ticket. Will you take a look and see if it is accurate and fits there or maybe needs moving elsewhere?

jofemodo, 2019-06-21:
Hi @riban-bw , it's OK like that. Thanks a lot!!

commit 7978791b (2019-06-20): Fix #74: Fix listbox up & down actions.