zynthian/zynthian-ui #80: Master MIDI Channel to Callable UI Action

jofemodo, 2019-04-11:
Map CUIA to MIDI Master Channel.
Implement CUIAs for zynswitches and selector up & down.

jofemodo, 2019-04-16:
Proposals for a better default note-mapping are welcome ;-)

riban-bw, 2019-04-21:
It would be advantageous to allow CC mapping as well as note mapping. That way one may use CC buttons  on their MIDI controllers whilst keeping the keyboard available for playing.

It would be advantageous to control the MIDI and Audio recorders (record / stop) from UIA. This would allow control of recording direct from the MIDI controller without menu navigation on the Zynth which can be irksome and interfere with one's artistic flow.

These two things combined would allow me to map the transport control buttons on my QX49 to control the recorder - perfect!

jofemodo, 2019-04-25:
A webconf configuration panel for CUIA events would be nice, but it's not a priority because most of users doesn't use CUIAs . Until then, you can use the rule system to map the default events to whatever you want, including CC ;-)

Anyway, i would accept a "better" note=>CUIA map ...

jofemodo, 2019-04-30:
I keep the issue open until a CUIA map configuration system is developed ;-)

jofemodo, 2019-11-26:
A map configuration tool has been already implemented for computer keyboards (thanks @riban-bw !) For MIDI we have only a fixed configuration, but it seems that nobody need more than this by now. If this were needed, we will open a new issue ;-)

closed 2019-11-26 by jofemodo