zynthian/zynthian-ui #81: Remove all effects can remove instrument instead

riban-bw, 2019-04-11:
Add a Synth Layer.
Add an effect to the synth.
In "Detailed Layer List" bold press on the effect.
Select "Remove All Effects".
Expected behaviour: The effect is removed.
Actual behaviour: The synth is removed.

jofemodo, 2019-04-12:
Yes. "Detailed layer list" is a "quick & dirty" implementation. It should be removed or improved ...

jofemodo, 2019-04-12:
OK! It should be fixed now ... although "detailed list" implementation is far from nice ;-)

jofemodo, 2019-04-14:
Could you confirm if it's solved for you?

Thanks!

riban-bw, 2019-04-15:
Sorry - this isn't working.

- From Layer page, create a NEW Synth Layer: FluidSynth: MIDI CH #1: FluidR3 GM: Yamaha Grand Piano
- From Layer page, bold press on synth and Add Effect: Auto phaser
- From Layer page, short press LAYER button to show Detailed Layer List
- From Detailed Layer List, bold press on 1#JV/Auto phaser: Remove Effect
Expected behaviour: Effect removed from chain.
Actual behaviour: Zynthian locks up with busy icon perpetually animated.

Running Zynthian from command line I see a warning when selecting "Remove effect":

> WARNING:root:zynthian_gui.zyncoder_read() => unordered types: NoneType() >= int()

jofemodo, 2019-04-15:
Try now ;-)

jofemodo, 2019-04-16:
Ups! Sorry! I didn't push the last commit on this ... i will do it tomorrow.

commit 85701b10 (2019-04-12): Fixes #81: Detailed list showing root layer options for sublayers.
commit fa38b7ab (2019-04-17): Fix #81: Last commit only solved the problem partially ;-)
closed 2019-04-17 by jofemodo