zynthian/zynthian-ui #82: Use symbols in status bar

riban-bw, 2019-04-12:
I would quite like to see symbols used in place of letters to indicate various status. Also use different colours to allow easier identification of message type (especially on smaller screens).

Use|Current | New | Unicode | Colour
----|---------|------|-----------|--------
Play | > | ► | 25BA | Green
Record | R | ● | 25CF | Dark Red
MIDI | M | ♫ | 266B | Blue
Over Temp | ▲° | 25B2, 00B0 | Bright Red
Under Voltage | ▼T | 25BC, 0056 | Bright Red

Would need to ensure each character renders appropriately in all available fonts. I don't know whether there is sufficient space to have two characters (as suggested for temperature and voltage).

It may be advantageous to change the border / background colour of a section of the screen (whole screen border or status area only) when an error occurs, e.g. X-run, over temp, under voltage, etc. This would differentiate from other fleeting but acceptable signals such as MIDI traffic or when a playback or recording starts or ends it can be difficult to differentiate the status change from coincident X-runs.

I can work up a PR for this if it is accepted as a desirable requirement.

wyleu, 2019-04-12:
That would be much appreciated! 
I would use a bright blue for under voltage... :-D so you could differentiate between the states with the ° and ≈ and hense use only one character.

jofemodo, 2019-04-12:
OK! It should be done. That's my preference:

Use|Current | New | Unicode | Colour
----|---------|------|-----------|--------
XRun | X | ✖ | 2716 | Bright Red
Play | > | ⏵ | 23F5 | Green
Record | R | ⏺ | 23FA | Dark Red
MIDI | M | ♫ | 266B | Blue
Over Temp | T | ℃ | 2103 | Bright Red
Under Voltage| V | ⚡ | 26A1 | Bright Red

jofemodo, 2019-04-12:
But if we want really cool icons, we should consider using something like FontAwesome:

https://fontawesome.com/icons?d=gallery

It can be downloaded as TTF file and used easily in the Zynthian-UI:

https://github.com/FortAwesome/Font-Awesome/tree/master/webfonts

;-)

Regards,

jofemodo, 2019-04-12:
This selection could fit our needs:

https://fontawesome.com/icons?d=gallery&c=audio-video,energy,weather

jofemodo, 2019-04-12:
OK! Already added FontAwesome. The font name for using from the UI code is FontAwesome ;-)

riban-bw, 2019-04-12:
Does this need to be added to webconf?

riban-bw, 2019-04-12:
I am not seeing this working in my updated zynth!

riban-bw, 2019-04-12:
PR #84 raised with changes to status. I had to tweak the position of the text to get the new font to work. It looks okay on my HDMI output at 320x280, 480x320, 800x600 and 1024x800 but less so on my tiny TFT at 160x128 - but that is going to be a challenge anyway.

I used icons that I think look okay - tell me what you think. I could not get the crotchets to display so maybe they are corrupt in the font pack. I used the same dark red for both MIDI and Audio record and for Record + Play as I used a different icon for this.

Sorry @wyleu I didn't use purple (as much as it is my favourite of the deep colours). I think usage will inform whether different icons / colours are required.

wyleu, 2019-04-13:
Sniff . . . .:-D

PR #84 (2019-04-14): Feature/status font
Fixes #82 - use icons / symbols in status bar.
commit 35ebf9d4 (2019-04-15): Fix #82: Adjustments to status bar icons.
closed 2019-04-14 by jofemodo