zynthian/zynthian-ui #83: Layer List doesn’t display correct preview name.

wyleu, 2019-04-12:
Layer List doesn’t display correct preview name.

IF I load up the Church bell from fluidsynth as a new engine then use the immensely useful preview in the GM 1 list to select fuzz guitar I will hear and play fuzz guitar but if I navigate to the Layer List the display still shows Church Bell.

If I select that entry it will reload Church bell, which means you have lost sight of the interesting sound you had found…

riban-bw, 2019-04-13:
If you select the previewed instrument then it will not be replaced by this workflow. If you just preview it , i.e. do not hit select button after scrolling to it then it will be replaced. I think that is logial, i.e. to change the selected instrument you need to select it first and the preview is just that, an ability to listen to instruments before selecting them.

wyleu, 2019-04-14:
But under that workflow it should revert to the sound that is selected once you move way from preview.. I have found.'dissinterested users' (musicians with no interest in implementation) are confused by this behaviour because they don't like to select a sound(I might damage something...) and they can hear the sound they want. It's just if they do select anything else (generally they want to record) they don't have the name of the preset displayed. To me it's the rule of least surprise... If the machine plays fuzz guitar I want the display to confirm that not tell me the preset I initially used to get into preview possibly 20 minutes ago.

jofemodo, 2019-04-14:
I will solve it ASAP ;-)

jofemodo, 2019-04-17:
Please, could you test if it's solved now?

wyleu, 2019-04-23:
Looks much better !!

commit 3908e12c (2019-04-17): Fix #83: Restore selected preset when clicking layer from preset screen.
closed 2019-04-23 by wyleu