zynthian/zynthian-ui #85: Fluidsynth preview does not change instrument until after note has been played

riban-bw, 2019-04-13:
Select a Fluidsynth layer.
(1) Scroll instrument list to an instrument, e.g."Rhodes EP".
Play a note.
Expected behaviour, highlighted instrument plays.
Actual behaviour, first note plays on previously selected instrument. Subsequent notes play on newly highlighted instrument.
Repeat from (1) - same result.

It seems that it is not just the first note that plays the "wrong" old instrument but the first set of concurrent notes, e.g. holding a chord will play all notes on old instrument then subsequent notes are played on newly highlighted instrument.

jofemodo, 2019-04-14:
Yes ... and it's not easy to solve. The system don't pre-load the presets until you play a note. As it takes some time to load the preset, this note is played with the last loaded preset...
What i could do is prevent the first note to be played ...
I also could try to preload the preset when you stay over a preset a minimum time, but this wont solve the problem if you are fast ;-) Also, some presets takes longer to load ...

jofemodo, 2019-04-16:
I will think about how to solve this problem, but unless i find a simple an elegant solution, i do nothing with it. Of course, proposals are welcome ;-)

riban-bw, 2019-05-21:
I think that preloading after a short (hover) period may be good but not if the load takes longer than a second or so. This would provide an uncomfortable interface where scrolling through the list might appear inconsistent.

I don't think the first key press should be ignored. Instead it should trigger loading of the preset (as per now) but not send the MIDI NOTE ON until the preset is loaded. If the the key has not been released then the NOTE ON should be sent once the preset is loaded.

Is there a mechanism for detecting when a preset is loaded? Such a mechanism is likely to be required for a reliable solution.

riban-bw, 2022-11-08:
This is a technically challenging issue. The current behaviour is wrong but slow loading instruments have unexpected behaviour in any mode. It is certainly obvious that the wrong sound should not play hence we should try to stop that from happening. Zynthian progress animation indicates that a preset is loading so there is visual feedback that something is changing.

Small improvement would be to not play the first note. Better would be to play the note (if it is still held) after the preset has loaded.

The first big challenge is that note on/off is sent directly to the engine by zynmidirouter and separately to the UI which handles preset change. Maybe the latter should be moved to the zynmidirouter codebase so that a decision can be made there on where and when to send the note-on message.