zynthian/zynthian-ui #86: Better use of screen space

riban-bw, 2019-04-13:
In many menus there is a block showing the zynthian icon which animates during busy periods and a block showing the menu name and the current selection, e.g. Layer / 2. This occupies about 1/4 of the screen and is mostly just diagnostic or temporal information, i.e. only required fleetingly. May I suggest that more space be given to the main menu text (which can be truncated on lower resolution screens / larger fonts) and this information be positioned elsewhere.

Examples:

- Position within a list (bottom right number) seems to only be diagnostic, useful for development. I recommend removing this.
- List name (bottom right text) may be presented (and already kind of is) in the topstrap. I recommend removing this.
- Zynthian icon seems to provide branding (which isn't necessarily required on all screens) and animates when the device is busy. My experience is that user input is ignored during such periods so it may be appropriate to present this as a popup activity indicator (like the hourglass or spinning wheel seen in desktop OS).

At the same time I would recommend changing the topstrap to use 1/8 rather than 1/10 of the vertical space which is more likely to be a factor of screen resolution and font size hence being less likely to truncate text. It also gives more space to the information most likely to be needed during performance, i.e. the current patch name.

If these measures were implemented then I think the display becomes more useful with more space for the elements the user wants to read without impacting usability.

Feel free to assign to me if you want it implemented.

jofemodo, 2019-04-14:
Really interesting. I'm thinking about it ...

jofemodo, 2019-04-18:
A little note ... the "zynthian logo" in the selector screens (right-top) acts as a "button" that you can click for accessing the snapshot screens. It's the only way of accessing snapshots when using a touch/mouse interface. Of course, it can be improved too ... ;-)

mheidt, 2019-04-18:
And is one of the crucial features I use the most on my touch-only zynthian.

jofemodo, 2019-04-18:
We could improve it, but not remove it  ;-)

riban-bw, 2019-05-05:
I have been trying to use the Zynthian with touch only for a day and find the experience inconsistent and a bit frustrating. Sometimes you have a back button and sometimes you need to find an unpopulated part of the top bar (which can prove challenging with long named patches). Sometimes attempts to swipe values up / down are interpreted as bold or long presses. I have previously read that the touch implementation is not the primary input method and hence is not optimised. If we have a touchscreen interface then I think it should have some attention paid to improve user experience. That may be the subject of a different ticket. Back to this topic... As a starting gambit - I suggest changing the scaling of the top bar to be /8 instead of /10. This multiple of 2 seems to make the top bar work better at various resolutions (which tend to be factors or powers of 2). I will create a PR / branch for this. (I also noticed some code optimisation around that bit.)

It may be appropriate to consider different screen layouts based on resolution / size / user preference / touch, etc. I wonder whether some effort to abstract the screen layout from the functionality might be advantageous? I can see a possible advantage of providing extra display where available, e.g. use of a SPI TFT plus HDMI output. (Oops - more scope creep. I must keep on topic!)

Let's consider whether there are any improvements that can provide touchscreen users a fantastic experience whilst avoiding excessive use of screen real estate.

@mheidt as a touchscreen user, how do you find the interface? Where do you see attention could be focused? How do you use the screen and avoid issues like I have found? Do you use a capacitive or resistive screen and do you resort to a pen or use your finger? My preference would be for a responsive touchscreen to allow reliable X-Y input. My experience tells me this is best achieved with capacitive screens.

mheidt, 2019-05-05:
I don't have issues pressing the back button.
I do have using rotary actions. Find my self too often I midi leaning.
But I find the official rotary encoders a pita as well. When I try to press, the selection shifts too easily.

I put the touch issue  on the shelf until the os supports multitouch. So I just live with it.

sn00zerman, 2019-06-26:
I on the other hand, find displaying the position within a list very usefull (I can easily remember numbers) I would find it a pitty to see this position dissappearing, in contrary, I would like to see something added to the "position display". Can you add a "total listcount" to this box ?
(so, if you have 50 items in the list, and you are on position 21, I would like to see not only "21", but would like to see this changed in "21 / 50".

Thanks in advance,
Kris

jofemodo, 2019-06-27:
Although it should be requested in a new ticket, i've conceded your desire...
Update and enjoy ;-)

riban-bw, 2019-06-27:
I think this info could be shown in a smaller part of the screen. On my HDMI display approx 1/8 of the screen is displaying this (tiny) number. I maintain that we should consider improvements to screen layout considering various screen size an resolution. Maybe something similar to Android profiles for small, medium and large displays.

jofemodo, 2019-06-27:
In the right time, everything will be improved. It's just a matter of priorities ...

sn00zerman, 2019-06-27:
Thanks for the quick response, but there is a small problem with this “total listcount”.
When I try to get the dexed list, it shows “/32”, but there are actually 320 items.
(I assume you do some formatting on the total listcount, to display 2 characters, while it actually needs 3 characters)

Best regards,
Kris

jofemodo, 2019-07-01:
Update and try now ;-)

jofemodo, 2021-07-05:
Currently this is being adressed by https://github.com/zynthian/zynthian-ui/issues/383.

closed 2021-07-05 by jofemodo