zynthian/zynthian-ui #87: MIDI Playback can leave sustain asserted

riban-bw, 2019-04-14:
If a .mid file has a sustain pedal controller message then after playback of the MIDI is stopped it is possible to leave the sustain pedal asserted which requires manual toggle of the sustain pedal to stop sustain. This is true of other controllers. It may be advantageous to reset some controllers at the end of MIDI playback.

riban-bw, 2019-05-10:
We could track controllers but that seems like a rabbit hole - could get lost.
We could send all_off on stop / end of track, assuming this clears controllers. This could impact other input, e.g. live playing. That might be acceptable. If this were a known limitation we could document that such scenario requires an extra event at the end of the .mid track to extend it long enough for any accompaniment, e.g. user adds a MIDI event 10s after last performance event in the .mid track. (We could even implement a feature to do this for the user but that feels like a step too far just now!)
If ALL_OFF does not fit the bill then we could see if MIDI RESET works or implement a forced reset function (similar to ALL_OFF).

mheidt, 2019-05-10:
https://github.com/zynthian/zynthian-ui/issues/66

jofemodo, 2020-03-10:
I think this is solved now. Please re-open if it's not ...

closed 2020-03-10 by jofemodo