zynthian/zynthian-ui #89: User interfaces cursor stops on dividers

riban-bw, 2019-04-16:
Each page in the UI has horizontal divider lines. It seems inappropriate for the cursor to stop on these lines as they can't be selected.

jofemodo, 2019-04-16:
You are right ... ;-)

jofemodo, 2019-04-17:
Try biw ;-)

riban-bw, 2019-04-17:
Hmmm! It doesn't stop at dividers but the behaviour seems less correct now! It still counts the divider so you have to cursor over it still. This feels less intuitive. I would prefer to see that you move directly from above the line to below the line when cursoring down.

jofemodo, 2019-04-18:
OK! I've improved a little bit the bahaviour, but the separator is still indexed internally as an item. Avoiding this is quite difficult because the separator is a real item in the listbox. Sorry ...

jofemodo, 2019-04-18:
FYI, what i implemented is a mechanism that skips separators and jump to the next item, detecting the direction of movement for adding or subtracting 1.

riban-bw, 2019-04-18:
Sorry - it is worse! Although it now skips past dividers, you can get stuck in the bottom block.
- Scroll down to the bottom of a page, past some dividers
- Scroll back up
Each time it reaches a divider it jumps back to the bottom.

riban-bw, 2019-04-18:
This looks like it may be related to zyncoder issue [12](https://github.com/zynthian/zyncoder/issues/12).

riban-bw, 2019-04-21:
zyncoder PR11 fixes my issue with getting stuck at bottom of list. (Error in the I2C HWC zyncoder lib.) Your implementation of this is elegant and simple. I really like it - well done! After zyncoder PR11 is merged I am happy to close this issue. Cheers

commit 296803f0 (2019-04-17): Fix #89: Separators in listboxes shouldn't be selected/highlighted.
closed 2019-04-25 by jofemodo