zynthian/zynthian-ui #91: ZynAddSubFX does not run with jack configured with 512 frames

riban-bw, 2019-04-16:
- Configure Jack with 512 frames using the -p 512 option
- Add news ZynAddSubFX layer
Zynthian hangs! This cause lots of delay due to ZynAddSubFX being my default patch so I could not start up Zynth to diagnose. Too little time left today to investigate further but wanted to report so it can be investigated.

mheidt, 2019-04-16:
https://github.com/zynthian/zynthian-ui/issues/65

jofemodo, 2019-04-16:
But 512 is a power of 2. It should be working fine ...
Although it's not a priority (default configuration with 256 works OK with all engines), it should be investigated ...

benru89, 2020-04-04:
I have found a solution for this, I know it is not at all your priority to have the onboard card working properly but maybe this problem can arise also with other configurations for other cards. ZynAddSubFx needs to be run with the same buffer size as jackd. ZynAddSubFx runs by defaut with a 256 buffer so it crashes when jackd is configured with another buffer size.
[image]
Adding -b 512 to the ZynAddSubFx initialization fixes it. Of course you would like to do it in a proper way, passing a parameter so it always matches what has been configured for jack.

jofemodo, 2020-04-07:
OK! It should be fixed now! Update and test, please...

jofemodo, 2020-04-07:
Thanks for the tip, @benru89 ;-)

commit f35066fe (2020-04-07): Fix #91: When starting ZynAddSubFX, setup sample-rate & buffer size to match those used by jackd.
closed 2020-04-07 by jofemodo