zynthian/zynthian-ui #92: New MOD layer hangs zynthian

riban-bw, 2019-04-18:
Add a new MOD layer:
- NEW Special Layer
- MOD-UI - Plugin Host

and the icon animates continually with no further user interaction possible. There are two errors at this point:

```
ERROR:root:Index out of range: 7
ERROR:root:list index out of range
```

This has recently been introduced. MOD previously worked on this device.

jofemodo, 2019-04-18:
It works for me now ...

riban-bw, 2019-04-18:
Not working here! Is there a way to reset the zynth to "factory settings" so that I we can baseline errors, e.g. a script that will reassert default (out of box) settings?

riban-bw, 2019-04-18:
I think this may be related to #91.  Different engines hang the UI under some conditions which may relate to specific combinations of engines or configuration. I currently have MOD working and am yet to figure out how to trigger this issue on demand.

jofemodo, 2019-04-19:
I don't think it's related with #91. #91 is reproducible, but "random freezes" in the UI are far more difficult to catch.

I'm thinking about integrating some kind of error reporting system in zynthian, so when the UI crash, the error trace is saved and when possible (next update), send-back to me for analysis. No stats, no user data, only error trace for catching hidden bugs.

What do you think about?

riban-bw, 2019-04-19:
I wouldn't regard this as random per se. When #91 or #92 occur, the problem persists until some change to the configuration, be that frame size, layers, etc. That is why I thought they may be related, the symptoms and behaviour is similar. Regarding error reporting, I would suggest avoiding automatic submission without user consent. It may prove challenging to ensure you comply with the data protection laws of every country in the world. Also, it may prove challenging to build a system that provides appropriate context without intruding on user data. A system that creates a fault report and allows the user to send it / approve submission / inspect content / whatever, may be more acceptable.

If you can detect a fault state then as well as capturing a report it works be good to recover, even if that is a reboot. Once the Zynth has hung out can be awkward to reboot it manually. 

In the meantime, let me know if there is any data I can collect during such a failure.

jofemodo, 2019-04-25:
Can you re-check if the problem is still there? Can you reproduce the error?

closed 2019-05-10 by jofemodo