zynthian/zynthian-ui #93: audio level indicator with peak display be able to replace the cpu bar

wyleu, 2019-04-22:
Would an audio level indicator with peak display be able to replace the cpu bar…?

If it switched to record level whilst recording that would be even sexier and a bit less nerdy . . .:smiley:

riban-bw, 2019-04-25:
I have a simple shared library which has two jack inputs and provides the maximum audio value since last query. It can provide left, right or both peak values. I have wrapped it in a Python wrapper and it seems to work. It does not (yet) have ballistics or logarithmic scaling but it may be sufficient to provide a simple thee state audio peak level indication, polled relatively slowly (like the same rate the status bar is updated). This could display a black / green / amber / red bar to represent four ranges of peak value.

riban-bw, 2019-04-28:
PR #97 adds a stereo audio meter that may be shown in place of the CPU meter. It needs some further work and collaboration but I think it is a reasonable start. Take a look.

jofemodo, 2019-04-29:
I merged PR#97 in the development branch gh93-levelmeter.
I commited some adjustments too ...

jofemodo, 2019-04-30:
OK Guys! It's merged with master!
Thanks a lot @riban-bw, I'm pretty sure that a lot of zynthianers will love because of this, including me, of course!! ;-)

jofemodo, 2019-04-30:
So, what about audio input DPM? Some notes:

=> Should we show 4 "bars" all the time or should we optimize for use cases:

+ If you are using zynthian as a stombox, most of times, only 1 input channel is used
+ If you are using zynthian as a stereo/mono recorder, only 1/2 input channels need to be metered. No need for output DPM.

=> What is the best layout/distribution for this? We could re-think the position of DPM bars ...

riban-bw, 2019-05-01:
@jofemodo will you take a look at this please? I spent (far) too long trying to figure out tkinter sizing! The meter assumes it has 'status_l' width but the status canvas does not scale linearly. On larger resolution screens the status canvas is narrower than status_l whilst on smaller screens it is wider than status_l. I tried forcing the canvas window size and getting the canvas window size but my efforts were futile.

wyleu, 2019-05-02:
The output bars are ALWAYS relevant because you might be combining any number outputs of audio engines and you ALWAYS want to know when you are over modding the output and hence clipping....

The input side is as you say option ridden. 
There would seem to be three engines required, since the zynthian is basically a steero beast ( it has two outputs at the moment) There are 
1/ A stereo input that is treated as one engine and brings in a stereo source . .
2/ A left mono input source which provides a nominally centred stereo image at -3dB so that it sums on output to the same signal volume but appearing as a centred stereo . This is probably what most people would expect a mono input to do which is appear in both speakers.
3/ the right equivalent of the above.

they would all have input gain control up and down from the nominal -18db Below mx modulation) so that a nominal reference tone would appear at the outputs as equivalent. . . . A pan control would allow the image to be positioned in a stereo. 

Presumably we could allow routing this so that a recording could be:

Channel 1 (L guitar) 
Channel 2 R microphone apres pre-amp to get it to line level) 

or: 

stereo output where guitar and vocals are positioned in their respective panned positions. 

The first would be of most use to subsequent editing, the later would b a performance mode.

riban-bw, 2019-05-02:
Regarding the last point, recording two channels and implementing free panning will allow for both scenarios, hard panned for dual mono or soft panned for stereo placement.

We could have four bars or we could allow toggling between input / output monitoring or how about toggling between:

- Stereo input
- Stereo output
- Both in & out (4 bars, each of half height)

We could provide a dedicated mixer page with level and pan control and level indication for each engine and input and outputs. I like the idea of a mixer with MIDI / OSC control. Such a page may reduce the need for simultaneous i/o monitoring in the status bar.

We could offer all options.

jofemodo, 2019-05-09:
Hi @riban-bw ! Just FYI:

https://github.com/x42/modmeter.lv2

Perhaps you can get some inspiration or trick from here ;-)

riban-bw, 2019-05-10:
Funny! I was looking at that yesterday too. It is an interesting array of meters which would be lovely to display on a dedicated meter bridge, e.g. second screen. I have not had a chance to play with it yet so have only seen the screenshots. I doubt we need jellyfish phase and level display for our little synth unless we see its future as a dedicated metering solution (as one of its multi-functions). I can imagine some of the meters take significant processing so may interfere with core audio function.

A colleague of mine created a dedicated audio monitor using Raspberry Pi which replaced very expensive monitoring units so this is certainly possible but maybe not on the same Pi as an audio engine.

I will install modmeter.lv2 and have a play to see what good ideas we can _borrow_. Cheers!

riban-bw, 2019-05-15:
Actually, I guess we should manage the scope of this request. We have delivered audio level meters that are currently fed from the main outputs. The request also asks for meters to show record level. @jofemodo please correct me if I am wrong but I believe that the built-in audio recorder is fed from the same point, i.e. main output in which case the meters already show record level. I suggest marking as resolved and asking @wyleu to review the implementation and close is done. We can revisit metering when we look at changes to the audio path.

wyleu, 2019-10-14:
Seems down to me to close.

closed 2019-10-14 by wyleu