zynthian/zynthian-ui #100: Moving layers to different MIDI channel

jofemodo, 2019-04-29:
From jerash:

change the midi channel from a layer, and change it to a channel already assigned to another layer. Then your synth is not listed anymore in zynthian-ui although it is receiving midi on the newly assigned channel and making sound.

jofemodo, 2019-04-29:
Solved by only allowing to move to free MIDI channels.

jerash, 2019-04-29:
ok, used channels do not appear anymore, so the problem won't come in anymore.
I've not used the "clone midi channel to" function yet, so I can't really argue if this could be an alternative as when you select an already used midi channel, then you could propose to clone the target midi chanel instead of changing it.

commit 7a824a31 (2019-04-29): Fix #100: When moving a layer to another MIDI channel, only allow free MIDI channels.
closed 2019-04-30 by jofemodo