zynthian/zynthian-ui #101: Audio level meter not always showing correct peak hold

riban-bw, 2019-04-30:
Increase audio until it is full scale 0dBFS.
Let audio fall back below full scale.
Increase audio to full scale.
Peak hold bars should be red at end but there are no peak hold bars.

riban-bw, 2019-04-30:
It looks like the status area extends beyond the width of the display so the peak hold bars are sometimes plotted outside the visible area.

riban-bw, 2019-05-01:
@jofemodo will you take a look at this please? I spent (far) too long trying to figure out tkinter sizing! The meter assumes it has 'status_l' width but the status canvas does not scale linearly. On larger resolution screens the status canvas is narrower than status_l whilst on smaller screens it is wider than status_l. I tried forcing the canvas window size and getting the canvas window size but my efforts were futile.

jofemodo, 2019-05-01:
I'm trying to reproduce the problem. What display sizes are causing the problem? How big is the "hidden" part of the DPM bars?

jofemodo, 2019-05-01:
OK. I understand ... let me see if i can solve it ;-)

jofemodo, 2019-05-01:
Try now, please ;-)

riban-bw, 2019-05-02:
Sorry - still not quite right. It fails on a screen with resolution 1024x600.  I found the issue was previously showing on larger resolution screens where the status area extended beyond the actual screen space. On smaller resolution screens the status area did not extend to the edge of the screen. I haven't had a chance to review your changes except to notice the casting to integer and setting variables during initialistation. (Good move - I left them in the loop for clarity so you could see what was happening. I don't know how much optimisation Python does but it felt wrong leaving them there!) I suggest trying on a variety of screen resolutions. Is there an easy way to create a constant tone, e.g. a sine wave for test purpose? (900 Hz - not 1kHz :-) ).

jofemodo, 2019-05-02:
Perhaps you didn't use the master branch? As the feature is already merged, i commited there.
Anyway, I have commited the changes to gh93-levelmeter and also integrated your last adjustments (range 30dB).

riban-bw, 2019-05-02:
I did use master branch. I now have a PD test tone so can investigate more acurately. I can see that the peak bar (brighter colour) does stop at the far right of the display but the peak hold does not seem to operate there. I will check for a fence post error in the library.

riban-bw, 2019-05-02:
Actually, the display is now fine but there is an issue whereby a signal that sustains at a level for longer than a few seconds then decays constantly, the bars won't show which is mostly desirable except at the top end where we want to see the over very clearly. I may look at a specific OVER indication to make this extremely obvious.

riban-bw, 2019-05-03:
Sorry @jofemodo! The canvas is still not perfect. On larger canvas the meter extends beyond the canvas and on smaller canvas it does not reach the end so the issue persists with scaling meter to canvas side. Meanwhile, there is also an issue with the peak library being too sensitive. I will add some hysteresis to the library. Could you look at the canvas again? I reckon it has to do with the sizer taking control which may need some hints (or brute force).

jofemodo, 2019-05-03:
Please, could you be more specific about display sizes?
Before commiting the changes, i've tested with 3 display sizes: 320, 480 & 1024 and it works perfectly in al of them.

riban-bw, 2019-05-04:
My bad! There were two things going wrong for me:

1. The window decoration on my x-server application was obscuring the edge of the canvas. (Stoopid Windows!)
2. In true Spinal Tap style, my library went up to 11 - well, it went over 1 (raw) or 0dBFS. That is okay except for physical outputs which should limit to 0dBFS. I have fixed this and also added hysteresis to the peak lib - expect a PR soon.

commit afa283d8 (2019-05-01): Fix #101: Audio level meter not always showing correct peak hold.
+ Some optimization and code refactorization.
closed 2019-05-04 by riban-bw