zynthian/zynthian-ui #103: Feature request: Master bus / insert on output

riban-bw, 2019-05-02:
It would be advantageous to have the concept of a master output bus with insert path. This may allow a master volume to be added which would be really useful and the ability to insert (effect) plugins in the output chain, e.g. a single compressor affecting all instruments.

The idea would be to mix all the instruments to the master bus then allow insertion of plugins between that mix point and the physical (or logical) output. (Actually, logical output makes sense, allowing the audio recorder to take advantage of this master bus chain.) The final "plugin" would be a level control. It may be possible to use an existing plugin that provides level control if it does not add to processing overhead, e.g. a limiter which does not consume CPU until activated. (I don't know if such exists.)

jofemodo, 2019-05-02:
Yes ... it seems to be the next step for bringing a full-featured FX-chain to zynthian ;-)

riban-bw, 2019-05-08:
I think the first step to this is to use an existing or create a new jack module that has 2 audio inputs, 2 audio outputs, mixes all connected inputs (jack does that already) and implements level control of the signal, i.e. OUTPUT_A = INPUT_A * LEVEL. LEVEL is a float between 0 - 1. This would then be permanently inserted before the system output (by auto router) and level meters will be connected to the output of this module. The module could be an LV2 plugin or a standalone Jack client (like jackpeak). When we have this we can then look at how we add / position LV2 plugins so that they may be in the audio input chain, individual instrument chain (already done) or output chain. That may be as simple as adding an extra menu option, e.g.
- NEW Synth Layer
- NEW Input Effect Layer
- NEW Output Effect Layer
- NEW Generator
- etc.

Or we could simply have the input and output layers permanently available so that a bold press could offer the context menu to Add Effect.

I can put together the jack client - should be quite quick. We could combine jackpeak to have a single module but that conflicts with the Unix ethos of one module for one function and it doesn't fit the chain - we want the monitoring after the output effects.

riban-bw, 2019-05-08:
I just played a bit and I think my preference would be to add a permanent JALV chain between the engines and system out, add a LV2 level control as the last element in this chain and have autoconnect ensure the meters are connected to the output of the chain (the level control plugin). Some module (e.g. autoconnect) would need to keep track of the output LV2 chain to know which is the first module and hence what to connect engines to. (I think that better / more efficient than having a dedicated mixer plugin at the start of the chain.)

It may take me a bit of time to get my head around the audio routing. @jofemodo would you want me to look at this or do you (or someone else) want to have a bash?

jofemodo, 2019-05-09:
Hi @riban-bw !

You understand the thing ;-) The functionality is already implemented with "FX layers", although we need a better UI and some routing-automation.

I will do it ASAP, but i would like to fix the current problems we have and release an official Aruk image. This is my current priority.

Thanks!

jofemodo, 2019-05-10:
I'm worried about performance. CPU is a very limited resource for Zynthian and some engines use almost all CPU power (Pianoteq, some ZynAddSubFX patches, etc.)

The global insertion FX-chain can be very convenient, but it's not needed at all for many use cases, so it should be something optional. Something like that:

1. Click "New FX layer" from the layer menu
2. Zynthian ask if you want to use it as:
   - Global FX-Chain (auto-route "other" layers output through it)
   - Audio Input FX-Chain (auto-connect to system audio input)
   - Custom FX-Chain (no auto-routing)

Also, we can auto-add a "Gain" plugin to FX-Chains, or have an option for that.

Regards,

riban-bw, 2019-05-10:
We could just have a "Master Bus" entry permanently in the Layer list. User can bold click to offer context menu like any other layer except cannot delete. Can manipulate plugin chain. If no plugins are in the chain then none are running and hence there is no processing. This removes the complexity of your menu workflow suggestion and makes it consistent with other workflows.

Regarding auto routing to the input of an effects layer, maybe we do something similar, i.e. all FX layers are created disconnected and we have an input "layer" from which we can route audio (or the opposite - all FX layers are auto routed but can be unrouted from the Input Layer).

I think a final gain would be advantageous when there is more than one plugin running as it allows overall level to be adjusted without then need to go into each layer and adjust its individual level. This would allow quick level reduction to avoid clipping.

I have drawn up some signal flow diagrams (and a mock up mixer screen which may be advantageous but the subject of a different topic). I hadn't grasped the audio routing concept until I read the source code then re-read the user docs. I think the audio signal flow diagrams may be advantageous on the wiki.

I think we can improve mono / stereo / dual mono routing but again - subject of a different topic.

jofemodo, 2020-03-27:
I've advanced quite a bit with this in a recent commit:

https://github.com/zynthian/zynthian-ui/commit/c8205671bb0048b5fb83d7294faaef175d98ed2d

Now you can create a pure FX-chain layer and use it as Master FX Bus, routing the output of other layers to the first element (or any) of this "Master FX Layer".

It's a simple step but it implements partially the features requested by this issue.

jofemodo, 2020-06-17:
I think this could be closed. We could talk about the UI & UX, but the functionality is already implemented.

riban-bw, 2020-07-02:
I don't think we do have the functionality yet. I appreciate it is _just_ a matter of sticking stuff together but we do not have that glue yet. I have created a 16 channel to stereo mixer which fits in with this request quite nicely (but is not a dependency) . You said you had some code to provide an output chain but that hasn't made it to master yet. I do think it would be advantageous to have the ability to insert a common audio chain. It is something that has been mentioned by several Zynthianers over the past few months who want an effect, e.g. reverb on several devices but can't afford the CPU / memory overhead to add to each engine.

jofemodo, 2020-07-02:
You don't need to put your FXs on every synth layer ...Try this:

+ Create a FX-layer in channel 15 and add the FXs you want in your Master FX Layer
+ Create your Synth layers normally and use the "Audio Output" option from layer options for sending the audio output to the channel's 15 layer.

Regards,

riban-bw, 2020-07-03:
Ah yes! I see. This is a technical solution which we may need to provide simpler user interface to. (I also need to tweak my zynmixer branch to accommodate this workflow.) I will close this issue and we can review how to provide a simpler user workflow as we move forward (hopefully with the mixer included :-) ).

closed 2020-07-03 by riban-bw