zynthian/zynthian-ui #104: Midi Router: Clone Midi does not clone Sustain

scblakely, 2019-05-03:
When I clone a midi channel from my keyboard to different channels (to create keyboard splits on different layers) the sustain messages are not cloned with the Note messages, so the additional sounds do not have matching sustain.

scblakely, 2019-05-04:
Having played around a bit more and thought about things a bit more, I think that a better option would be to implement a CLONE command in the midi router (as an alternative to MAP) so users could CLONE specific midi messages from a channel to a new channel for a different layer.

That way I could CLONE the sustain CC message from the main keyboard channel to the additional layers.

jofemodo, 2019-05-06:
The reason for creating the "clone" feature and disallowing the possibility of having several engines in the same MIDI channel was that normally you want to change CCs independently for every engine, although you play unison. Obviously, depending on the use case, you could want to clone some CCs too.

The "clone" operator in the MIDI router is a good option for those cases, but perhaps we could find a more "user-friendly" solution ... let's think about it a little bit ;-)

jofemodo, 2020-02-10:
Implemented on:

+ https://github.com/zynthian/zyncoder/commit/98ff2ae0ddaea97f7aa42a5f2c23c15404f61da1
+ https://github.com/zynthian/zyncoder/commit/e3218b24337834036c9a0d2c5b442a1b5e12e978
+ https://github.com/zynthian/zynthian-ui/commit/aeb0d7f810937110fcda0d5a4cd491e550a1a3e4

closed 2020-02-10 by jofemodo