zynthian/zynthian-ui #105: Midi Profile needs UI restart

mheidt, 2019-05-04:
I haven't tested with the latest build. I am in concert mode and won't update.
I have 2 Midi Profiles with midi rules.
When I change the profile, it is only taken into account, when i restart the UI

mheidt, 2019-05-05:
Even on an updated image, this bug still occurs.

jofemodo, 2019-05-07:
Do you change the MIDi profile from webconf or from UI?
I just tested with a Gorgona Omega totally updated and it worked for me when changing from both but when using rules, then i noted that rule table is not conrrectly reseted when changing and rules remain from a MIDI profile to another. Could you confirm this behaviour as your problem?

jofemodo, 2019-05-07:
I just commited some changes that should solve the "rules reseting" problem.

jofemodo, 2019-05-07:
**NOTE:** MIDI-learning could do strange things when deleting or adding the profile's script rules. 
You have to think that native MIDI learning is implemented using rules that sometimes are chained when learning, so deletig or adding new rules could change the MIDI-learning map.

mheidt, 2019-05-08:
I can confirm,that it  works for me now.
What about saving the midi profile in the snapshot as well?

jofemodo, 2019-05-09:
I like the idea of saving MIDI profile inside snapshots.
Please, open a new task for that ;-)

commit 90a520fb (2019-05-07): Fix #105: Remove MIDI profile script's rules before loading a new MIDI profile.
+ Also ends properly lib_jackpeak + minor fixes
closed 2019-05-09 by jofemodo