zynthian/zynthian-ui #106: Cannot load Noize Mak3r - Plugin LV2

riban-bw, 2019-05-05:
From Layer List:
- REMOVE ALL
- NEW Synth Layer
- Noize Mak3r - Plugin LV2
- Expected behaviour: Able to select MIDI Channel
- Actual behaviour: loading icon animates continually.

Output from terminal:
```
ERROR:root:Can't start engine Jalv/Noize Mak3r => End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x74ccead0>
command: /usr/local/bin/jalv
args: ['/usr/local/bin/jalv', 'http://kunz.corrupt.ch/products/tal-noisemaker']
buffer (last 100 chars): b''
before (last 100 chars): b'k length: 256 frames\r\nMIDI buffers: 32768 bytes\r\nComm buffers: 524288 bytes\r\nUpdate rate:  25.0 Hz\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 1719
child_fd: 20
closed: False
timeout: 20
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile(b'\n> ')
ERROR:root:Can't exec engine command: \info_controls => End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x74ccead0>
command: /usr/local/bin/jalv
args: ['/usr/local/bin/jalv', 'http://kunz.corrupt.ch/products/tal-noisemaker']
buffer (last 100 chars): b''
before (last 100 chars): b'\\info_controls\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 1719
child_fd: 20
closed: False
timeout: 20
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile(b'\n> ')
ERROR:root:Can't exec engine command: \get_presets => End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x74ccead0>
command: /usr/local/bin/jalv
args: ['/usr/local/bin/jalv', 'http://kunz.corrupt.ch/products/tal-noisemaker']
buffer (last 100 chars): b''
before (last 100 chars): b'\\get_presets\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 1719
child_fd: 20
closed: False
timeout: 20
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile(b'\n> ')
Set up encoder i=3, pin_a=4, pin_b=0, midich=0, midictl=0, oscpath=(null), value=0, maxval=60, step=1
ERROR:zynautoconnect.zynthian_autoconnect:'NoneType' object has no attribute 'encode'
```

jofemodo, 2019-05-06:
What image are you using? Is it the RC1 or Gorgona Omega?
For me it's working in both SD images ...

jofemodo, 2019-05-07:
Hi @riban-bw !
I can't reproduce this problem in the "official" images ...

jofemodo, 2019-05-09:
Hi @riban-bw !

If you can't add more info to that, i would like to close. I can't reproduce ...

riban-bw, 2019-05-09:
This is happening with Gorgona Omega updated today. I will download Aruk RC1 and test on that. It happens every time.

jofemodo, 2019-05-10:
I can't reproduce on Gorgona (updated). Could you confirm with a fresh Gorgona+update?

Thanks!

riban-bw, 2019-05-10:
- Fresh install of Gorgona
- Set date
- Update
- Configure Audio: Hifiberry Dac+light, -P 70 -t 2000 -s -d alsa -d hw:0 -r 44100 -p 256 -n 2 -X raw
- Configure Display: Custom Device, /dev/fb1
- Configure Wiring: I2C Hwc
- Load kernel modules: spi_bcm2835, fbtft_device
- options fbtft_device name=sainsmart18 rotate=90
- UI Options: Font=Helvetica 10pt.
- Another Update
- Add synth layer setBfree
- Test sound okay
- Remove all layers
- Add synth layer Noize Mak3r
ERROR:root:Can't start engine Jalv/Noize Mak3r => The command was not found or was not executable: /usr/local/bin/jalv.
ERROR:root:Can't add layer Noize Mak3r - Plugin LV2 => 'NoneType' object has no attribute 'split'
- Update again
Same result

This is worse - jalv is not installed. What is going on???

jofemodo, 2019-05-12:
The update didn't work as expected.
You could try to update again or start from a fresh Gorgona.
If you do the last, then you should update from webconf or CLI and save the logs ;-)

Perhaps Gorgona is too old and htpdate is having problems for updating the system date/time...

riban-bw, 2019-05-12:
I'm not sure what went on there. I have now installed Aruk RC-1 and it works (until JACK stops hackling). I will close this issue.

closed 2019-05-12 by riban-bw