zynthian/zynthian-ui #107: Qjackctl not displaying

riban-bw, 2019-05-06:
- ssh to Zynthian
- qjackctl

qjackctl seems to be running but the windows do not render in my x-server. Same if it point at a different x-server, e.g.
DISPLAY=192.168.1.8:0 qjackctl.

On my Windows PC running VcXsrv (which works for other applications forwarded from the zynthian) I see two windows (default and connections) in the task bar but neither will render on the screen.

jofemodo, 2019-05-06:
Are you using a ZynthianOS image? Please, confirm ;-)

riban-bw, 2019-05-06:
I can't remember but probably Gorgona. Is there a way to tell?

jofemodo, 2019-05-06:
Yes. The OS line in the webconf dashboard. If it's "Jessie", then it's a Gorgona. If it's  "Stretch", then is Aruk ;-)