zynthian/zynthian-ui #109: Setting QMidiNet in UI has no effect

riban-bw, 2019-05-08:
- Navigate to Admin menu
- Toggle QMidiNet setting
- Restart UI
Expected behaviour: QMidiNet setting is retained.
Actual behaviour: QMidiNet setting returns to previous value.

It seems like this setting may only be set from the web-conf.

jofemodo, 2019-05-09:
Yes. Currently Qmidinet flag status is not saved when modified from UI.
But it works! Aka, QMidiNet is correctly started/stopped.

jofemodo, 2019-05-09:
I made "MIDI Network" option to be persistent (saved in MIDI profile) when changing from UI.
BTW, 2 new MIDI options were added to UI's Admin menu (also persistent!):

 - midi_prog_change_zs3
 - preset_preload_noteon

jofemodo, 2019-05-13:
Test this, please ;-)

closed 2019-05-14 by jofemodo

riban-bw, 2019-05-16:
The state is now persistently changed by UI. I don't use MIDI Network so assume it continues to work as expected.

commit f0839eac (2019-05-09): Fix #109: Make "MIDI Network" option to be persistent (saved in MIDI profile) when changing from UI.
+ BTW, add 2 new MIDI options to UI's Admin menu (also persistent!): midi_prog_change_zs3 & preset_preload_noteon