zynthian/zynthian-ui #110: Listbox truncated

riban-bw, 2019-05-08:
On a large resolution screen, e.g. 800x600 the Layer List (and others) truncates the text in the listbox due to the width being too narrow. The centre half of the screen is unpopulated so there is plenty of space to expand this listbox width.

riban-bw, 2019-05-08:
This seems due to `self.lb_frame.columnconfigure(0, weight=10)` being commented out in zynthian_gui_selector.py.

jofemodo, 2019-05-09:
What font size are you using? For 800x600 a font size of 24 should be fine.
I've tested and it works quite well for me, although some pixel is lost in the bottom ...

Anyway, scaling *perfectly* to fit any display size is not a priority for me. Of course, if you want to try to improve it, you are very welcome!!  ;-)

riban-bw, 2019-05-09:
This isn't an issue with font size. It occurs on any resolution screen. The problem is that the code to allow the listbox to span the allocated space is commented out and hence, although the frame extends across 3/4 of the screen, the listbox remains limited to 1/4 of the screen. I wonder why this line was commented. It seems a simple fix to remove the comment # and then the application behaves as expected with the full width available for the listbox. I will gladly create a branch (or include it in the gh86-screenspace branch changes) but it is only one character change and I wanted to understand why the line was disabled. (It is line 170  of zynthian_gui_selector.py in the current master branch.)

jofemodo, 2019-05-10:
OK! You seem to be right ;-)
I can't remember why this line is commented and don't waste my time doing a complete research. I've tested and can't see any side effect when enabling it, and the listbox is correctly expanded, so i'm committing the change and we stay alert for any side effect we haven't detected.

commit 9c58d158 (2019-05-10): Fix #110: Listbox not expanding correctly. Uncomment "columnconfigure" line solves the problem.
closed 2019-05-10 by jofemodo