zynthian/zynthian-ui #111: MOD-UI don't save plug-in presets

rodamaralgeo, 2019-05-08:
How to reproduce the error:
1) Start Zynthian
2) Add a Layer with MOD-UI
3) Add an LV2 plugin
4) Save a preset in the plugin
5) Restart Zynthian and verify that the preset has been saved

jofemodo, 2019-05-09:
That work for me, although is a little bit confusing because the issue #112 .
When you save the pedalboard, it's saved with a different name and the saved presets are there.
I think that when #112 is solved, this issue will be solved too ;-)

jofemodo, 2019-05-10:
NOTE: You have to save the pedalboard for getting the pedalboard's presets really saved on it.

rodamaralgeo, 2019-05-11:
Hi @jofemodo 

I made two videos showing the problem

1) Saving the prestes seems to be working everything correctly

https://drive.google.com/file/d/1Z3blbClkqbkXP4gVa9ZAYptmC6eueWFV/view

2) After restarting Zynthian, the presets are not listed

https://drive.google.com/file/d/1PjlNz7yQU2JfdcBrvexFpP35wDluHBWO/view

Note: I have already change the MOD-UI branch

jofemodo, 2019-05-13:
I succesfully saved 3 presets in a pedalboard and the 3 presets were there after rebooting the zynthian.
Did you save the pedalboard itself before rebooting?

rodamaralgeo, 2019-05-14:
When i save the preset everything seems ok, but when I try to load the saved preset (without restarting zynthian) it is not loaded.

When trying to load the preset I get the following error:

> [I 190514 23:05:14 web:1825] 200 GET /effect/preset/save_new//graph/_3BandEQ_1?name=Preset+1&_=1557871317715 (192.168.0.102) 9.42ms
[I 190514 23:05:15 web:1825] 200 GET /ping?_=1557871317716 (192.168.0.102) 3.08ms
[I 190514 23:05:16 web:1825] 200 GET /ping?_=1557871317854 (192.168.0.102) 2.95ms
[I 190514 23:05:20 web:1825] 200 GET /ping?_=1557871317717 (192.168.0.102) 4.78ms
[I 190514 23:05:22 web:1825] 200 GET /ping?_=1557871317855 (192.168.0.102) 3.83ms
[E 190514 23:05:23 web:1421] Uncaught exception GET /effect/preset/load//graph/_3BandEQ_1?uri=file%3A%2F%2F%2Froot%2F.lv2%2F_3BandEQ_1-Pr                              eset_1.lv2%2FPreset_1.ttl&_=1557871317718 (192.168.0.102)
    HTTPServerRequest(protocol='http', host='zynthian.local:8888', method='GET', uri='/effect/preset/load//graph/_3BandEQ_1?uri=file%3A%2                              F%2F%2Froot%2F.lv2%2F_3BandEQ_1-Preset_1.lv2%2FPreset_1.ttl&_=1557871317718', version='HTTP/1.1', remote_ip='192.168.0.102', headers={'Re                              ferer': 'http://zynthian.local:8888/?v=1557871194', 'X-Requested-With': 'XMLHttpRequest', 'Connection': 'keep-alive', 'User-Agent': 'Mozi                              lla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/74.0.3729.131 Safari/537.36', 'Accept-Encoding': 'gzi                              p, deflate', 'Host': 'zynthian.local:8888', 'Accept': 'application/json, text/javascript, */*; q=0.01', 'Accept-Language': 'pt-BR,pt;q=0.                              9,en-US;q=0.8,en;q=0.7'})
    Traceback (most recent call last):
      File "/usr/local/lib/python3.5/dist-packages/tornado/web.py", line 1302, in _stack_context_handle_exception
        raise_exc_info((type, value, traceback))
      File "<string>", line 3, in raise_exc_info
      File "/usr/local/lib/python3.5/dist-packages/tornado/stack_context.py", line 314, in wrapped
        ret = fn(*args, **kwargs)
      File "/usr/local/lib/python3.5/dist-packages/tornado/gen.py", line 131, in final_callback
        if future.result() is not None:
      File "/usr/local/lib/python3.5/dist-packages/tornado/concurrent.py", line 209, in result
        raise_exc_info(self._exc_info)
      File "<string>", line 3, in raise_exc_info
      File "/usr/local/lib/python3.5/dist-packages/tornado/gen.py", line 810, in run
        yielded = self.gen.throw(*sys.exc_info())
      File "/home/pi/zynthian-sw/mod-ui/mod/webserver.py", line 862, in get
        ok  = yield gen.Task(SESSION.host.preset_load, instance, uri)
      File "/usr/local/lib/python3.5/dist-packages/tornado/gen.py", line 807, in run
        value = future.result()
      File "/usr/local/lib/python3.5/dist-packages/tornado/concurrent.py", line 209, in result
        raise_exc_info(self._exc_info)
      File "<string>", line 3, in raise_exc_info
      File "/usr/local/lib/python3.5/dist-packages/tornado/gen.py", line 491, in Task
        func(*args, callback=_argument_adapter(set_result), **kwargs)
      File "/home/pi/zynthian-sw/mod-ui/mod/host.py", line 1508, in preset_load
        self.send_modified("preset_load %d %s" % (instance_id, uri), host_callback, datatype='boolean')
      File "/home/pi/zynthian-sw/mod-ui/mod/development.py", line 72, in send_modified
        callback(True)
      File "/home/pi/zynthian-sw/mod-ui/mod/host.py", line 1506, in host_callback
        self.send_notmodified("preset_show %s" % uri, preset_callback, datatype='string')
      File "/home/pi/zynthian-sw/mod-ui/mod/development.py", line 77, in send_notmodified
        callback(True)
      File "/home/pi/zynthian-sw/mod-ui/mod/host.py", line 1481, in preset_callback
        for symbol, value in get_state_port_values(state).items():
      File "/home/pi/zynthian-sw/mod-ui/modtools/utils.py", line 650, in get_state_port_values
        values = structPtrToList(utils.get_state_port_values(state.encode("utf-8")))
    AttributeError: 'bool' object has no attribute 'encode'
[E 190514 23:05:23 web:1825] 500 GET /effect/preset/load//graph/_3BandEQ_1?uri=file%3A%2F%2F%2Froot%2F.lv2%2F_3BandEQ_1-Preset_1.lv2%2FPr                              eset_1.ttl&_=1557871317718 (192.168.0.102) 57.45ms
[E 190514 23:05:23 web:1825] 500 GET /effect/preset/load//graph/_3BandEQ_1?uri=file%3A%2F%2F%2Froot%2F.lv2%2F_3BandEQ_1-Preset_1.lv2%2FPr                              eset_1.ttl&_=1557871317718 (192.168.0.102) 60.43ms

jofemodo, 2019-05-30:
It works perfectly for me:
+ Add a MOD-UI layer
+ Create some pedalboard presets
+ Save the pedalboard
+ When rebooting, the presets are available.

rodamaralgeo, 2019-06-01:
Pedalboards presets are working perfectly.

The reported problem occurs when I try to save presets for each of the lv2 plugins that are on pedalboard

jofemodo, 2019-06-03:
Ups!! Bad reading ... i will check again ;-)