zynthian/zynthian-ui #112: MOD-UI Pedalboards are being copied when saved

rodamaralgeo, 2019-05-08:
How to reproduce the error:
1) Start Zynthian
2) Add a Layer with MOD-UI
3) Save the Pedalboard
4) Restart Zynthian, edit Pedalboard and save
5) Check the list of Pedalboards

[image]

[image]

jofemodo, 2019-05-09:
Hi @rodamaralgeo !

Are you using the MOD-UI version included on zynthian or you have updated? ;-)

Regards,

mheidt, 2019-05-09:
I fixed this issue 2 years ago.
https://github.com/zynthian/mod-ui/pull/1
Are you sure, that you have this change in your branch?

rodamaralgeo, 2019-05-09:
I downloaded and installed the ZynthianOS image from 2019-05-03

The only thing done in the installation was the Zynthian update

steveb, 2019-05-09:
I've seen this too, and I haven't figured out how to reliably reproduce it. Something like loading the pedalboard from zynthian-ui then saving it from mod-ui

jofemodo, 2019-05-10:
Hi @mheidt !

I've inspected the code and your change (commit 030219a5c6362e690353d52fbd864abba4f7dfa3) is missing. Probably it got lost by @guysoft while compacting all "zynthian" changes in 1 single commit. 
Or perhaps it was me while merging ... ;-) If somebody want to investigate, he could start by here:

https://github.com/zynthian/mod-ui/commits/master/mod/host.py

I've tried but i can't trace where the change is lost ... although i would like to understand for avoiding similar problems in the future.

Anyway, i'm restoring the lost commit ;-)

jofemodo, 2019-05-10:
OK! Now i understand the problem. The commit is not lost in master, but it's lost in the "zynthian-single-commit". For some reason, @guysoft changed the branch used on mod-ui install script:

https://github.com/zynthian/zynthian-sys/commit/b34cc88ca7e4778db3d3d8b06b13a1f0b021c88f#diff-034b5e27cf7cd98185b05a4027eb9e56

And current ZynthianOS images has this branch enabled by default. You can change it like this:

```
cd /zynthian/zynthian-sw/mod-ui
git checkout master
```
I've changing the mod-ui install recipe to use the master branch again ...

rodamaralgeo, 2019-05-11:
Hi @jofemodo 

After change mod-ui banch works fine!

Thanks!

guysoft, 2019-05-11:
@jofemodo Hey, I merged it with your approval.
Here is the talk about that, might jog your memory :)
https://github.com/zynthian/zynthian-sys/issues/55#issuecomment-407714066

jofemodo, 2019-05-12:
Hi @guysoft !
There is no problem with the merged master, but the development branch "zynthian-single-commit" missed some commit. The problem is that the mod-ui install recipe was changed to use this development branch instead of the master branch. I can't remember why this change was done, but anyway, it's not so important right now because it's already solved. We will try to improve our proccesses for avoiding this kind of problems in the future ;-)

closed 2019-05-13 by jofemodo