zynthian/zynthian-ui #113: MIDI profile / snapshot

mheidt, 2019-05-09:
Save MIDI profile in snapshot.

jofemodo, 2019-05-09:
We should save the current MIDI-profile state or the MIDI-profile name?
What do you think is better?

mheidt, 2019-05-09:
Very good question.
But I would expect the right selection in the profiles list as well. So I would go with he name.
And if there is no profile with that name, take Default.
If you took the state, you don't have anything to select. And selecting default and using a different state is a bug.

jofemodo, 2019-05-09:
I could save the state + name, restoring the state and selecting the name, although saved state doesn't fit exactly with the named profile, that could have been modified... In that case, snapshot's MIDI-profile state will load instead of name's one.

mheidt, 2019-05-09:
And if name doesn't exist you will persist that new profile which exists even when snapshot isn't loaded anymore?
I like it

jofemodo, 2019-06-27:
OK! What should we do when loading a snapshot with a given MIDI profile, and after that, remove all layers for starting from zero. Should we come back to the "system configured" (saved in config) MIDI profile?

mheidt, 2019-06-27:
I don't quite understand. You wanted to save the values and the name of the profile.
If the profile name doesn't exist, you would create the new profile...for ever...
And if the profile exists, we don't care, if the values are the same. You overwrite the values taking from the snapshot.

closed 2019-11-15 by jofemodo