zynthian/zynthian-ui #114: Fails to connect audio to I2S output if USB soundcard inserted during boot

riban-bw, 2019-05-11:
- Configure Zynthian to use I2S audio interface, e.g. Hifiberry Dac+ Light
- Plug class compliant USB audio interface to USB port
- Boot Zynthian
Expected behaviour: Audio routed to I2S audio interface.
Actual behaviour: Audio routed to USB audio interface.

jofemodo, 2019-05-12:
Yes. I've tried lot of times to force the correct ordering of audio devices, but with no luck. It seems to not work or i'm not understanding the ALSA documentation, etc.

The issue also prevents Zynthian to work correctly when booting with several USB controllers plugged in. This is, from my POW, much more important for most of use-cases.

jofemodo, 2021-07-25:
I think this is solved by using explicit device names on jack configuration instead of device numbers.

closed 2021-07-25 by jofemodo