zynthian/zynthian-ui #115: Unplug USB audio interface breaks Zynthian

riban-bw, 2019-05-11:
- Start Zynthian with USB audio interface plugged in
- Remove USB audio interface
Expected behaviour: Zynthian UI continues to be responsive, e.g. allows menu navigation, reboot, etc.
Actual behaviour: Zynthian UI becomes unresponsive. Operating system continues to function but would require hard power reset if ssh / console access is unavailable (which is the normal operating condition).

Although one might expect Zynthian to be of limited use if it loses its audio interface, the USB connector is not the most resilient and there is the possibility of it becoming dislodged at which point one would expect to be able to recover without hard power reset of Zynthian. Also, any unfinished work will be lost.

jofemodo, 2019-05-12:
The kind of robustness you describe here is difficult to achieve and USB device support is not a priority for me. Anyway, if somebody find a good method for achieving it, she's very welcome to implement it ... ;-)

jofemodo, 2021-07-25:
When unplugging an USB audio device that is used for audio, jack dies and this cause that all jack-related stuff fail, including, among others:

+ MIDI router (zyncoder)
+ Peak meter (jackpeak)
+ Autoconnector
+ All running engines

So, it's really difficult to bring this kind of robustness to zynthian UI. UI can't start without jack, so it can't continue working without jack.

The only thing we can do is detecting jack i not running and keep trying to restart the UI until jack is running again. This shouldn't be too difficult to implement.

Regards

riban-bw, 2021-07-25:
When the UI is separated from the core audio modules then we should implement a watchdog which checks health of audio and restarts / reconfigures jackd as required. We may want to add some fallback rules, e.g. use default soundcard if configured soundcard disappears.