zynthian/zynthian-ui #116: Feature request: Encoder test screen

riban-bw, 2019-05-17:
It may be advantageous to be able to test the functionality of the hardware. This will assist release testing and also provide a diagnostics tool. Proposed implementation:

- Navigate to Admin menu
- Select (new) menu item "Hardware Test"
- (new) screen displayed showing values for each encoder
- Rotate an encoder changes associated on-screen value
- Short press encoder indicates short press on screen (maybe with colour and / or words)
- Bold press encoder indicates bold press on screen (maybe with colour and / or words)
- Long press encoder indicates long press on screen (maybe with colour and / or words)
- Long press BACK encoder presents "Exit Test - Yes / No" screen
- Future enhancement may add list of hardware to test that may be navigated with SELECT encoder.

wyleu, 2019-05-18:
I would widen this to MIDI IN as well, it's a rather work doesn't work sort of thing and if you fall into the doesn't work there's not too much you can do to diagnose the issue.

jofemodo, 2019-11-15:
Although it's not exactly the same, i've included a little "auto-detection" code for the MCP23017 in the webconf's dashboard. So, now you can know it the system is detecting the GPIO expander in the I2C bus ...