zynthian/zynthian-ui #117: x-runs with Pianoteq using default configuration

riban-bw, 2019-05-18:
Out of the box Aruk configured for Hifiberry DAC+ Light x-runs every few seconds.

JACK config: -P 70 -t 2000 -s -d alsa -d hw:0 -r 44100 -p 256 -n 2 -X raw

We should provide a better experience, maybe with different settings.

Turning off effects improves slightly but there are still several x-runs each minute. x-runs are significantly reduced by setting internal sample rate to 11025Hz (which results in dull sound) or setting JACK to use -p 512 which increases latency. Neither is ideal but x-runs are far worse.

jofemodo, 2019-05-20:
It works normally for me. Depending of the selected instrument, you can get XRUNS when playing hard, but it's been like that from the very beginning.

jofemodo, 2019-11-26:
RBPi4 solves the problem ;-)

commit 28fd7db3 (2020-03-10): Fix #117: MIDI Looping back on usb devices. Fix MIDI autoconnector.
closed 2019-11-26 by jofemodo