zynthian/zynthian-ui #118: Engines may buzz when being removed

riban-bw, 2019-05-18:
If an engine is creating audio when it is removed from a layer then JACK continues to send audio to the output for a while resulting in a buzz on the output. It would be advantageous to mute the engine or unroute it from output before destroying the engine.

wyleu, 2019-05-19:
Good Observation

closed 2019-05-20 by jofemodo

jofemodo, 2019-05-22:
Could you test this?

riban-bw, 2019-05-22:
Yes, I tested when I saw the issue was closed and it seems to work nicely thanks. It would be good to have a resolved state for issues, i.e. issue is fixed awaiting test. We would expect the original reporter to test and close issue. I'm not sure how best to implement that in GitHub.

jofemodo, 2019-05-22:
The canvas rules ;-)
We have "testing" and "Done" columns ;-)

commit a0a3b45f (2019-05-20): Fix #118: Mute layer before deleting for avoiding ugly noises.