zynthian/zynthian-ui #122: MIDI Recorder playback fails after some time

riban-bw, 2019-05-20:
MIDI playback on loop stops output after some hours.

**Steps to reproduce:**
- Add ZynAddSubFX engine to MIDI channel 1 with Organ 1 patch
- Navigate to Admin mneu ->MID Recorder
- Enable Loop Play
- Start playback of a MIDI file
- Leave for several (up to 9) hours
Expected behaviour: Audio continues to play for minimum of 24 hours*
Actual behaviour: Playback appears to be ongoing (green play arrow, processes running) but no audio output and no level on meters.

24 hours is an arbitrary duration chosen for minimum resilience / uptime. 

**Setup:**
 - Aruk RC-1 2019-05-17-zynthianos-stretch-lite-1.0.0
 - RPi3B+, I2S PCM5102A DAC Hifiberry DAC+ Light, HDMI displa, I2C HWC
 - zyncoder: master (f672128) 
 - zynthian-ui: master (3ae0262) 
 - zynthian-sys: master (92a6093) 
 - zynthian-data: master (6712b1c) 
 - zynthian-webconf: master (cca23d2)
 - No USB MIDI devices connected

It looked like all the modules were still running and JACK routing remained as expected (from jack_lsp -c) but the player MIDI output had stopped or was not reaching the synth engine.

Playing external MIDI keyboard did produce sound so the synth engine was still functioning.
There was no blue "m" indicating that Zynthian was not receiving MIDI from the player.

jofemodo, 2019-05-21:
How much time did you have to wait to see the error? 9 hours?
Probably it's problem with the player. I imported the source code to github and made some changes for having the "loop" feature:

https://github.com/zynthian/jack-smf-utils

It should be some  "counter" problem in some place. We should run the player standalone and see if there is some error message or it's a "segmentation fault".

Regards,

riban-bw, 2019-05-21:
I left this running over night (to perform an availability test) and discovered the issue in the morning, approx. 9 hours after starting the test. Whilst putting together some test cases I included up-time and availability and used the Audio and MIDI Recorder loop playback features to test the unit. I chose 24 hours as a duration that I thought might be appropriate for a synthesizer - it is much longer than most normal use cases (performance sessions) but may be exceeded if the unit is used for installation art, etc. A separate stress test would identify the actual longevity of the product. Of course, if the testing app fails then we can't test the rest of it!

riban-bw, 2019-05-21:
I was able to reproduce this issue today having started jack-smf-player running at 06:16:14 playing to Fluidsynth. When I checked at 19:00 the same condition existed, i.e. jack-smf-player still running but no MIDI being triggered. I manually connected jack-smf-player to Fluidsynth using jack_connect and this did not trigger any MIDI. USB MIDI input does trigger Fluidsynth.

Journal messages were being logged:

```
zynthian startx[1344]: jack-smf-player: Metadata: End Of Track

```
The last journal message was at 12:05:48 showing it ran for approx. 20974s before MIDI was muted.

riban-bw, 2019-05-22:
Attempt to restart jack-smf-player will appear to succeed but actually give no MIDI output until RPi is rebooted. I have started a manual instance of jack-smf-player running to see if there are any extra messages when it stops working.
7 hours later and it is still running. Only one experiment but first result suggests problem may lie within python hosting jack-smf-player.

jofemodo, 2019-05-22:
Still working?

riban-bw, 2019-05-22:
I was running in a terminal on my laptop which needed to be closed for transport. I should have run out in a screen session. I can't connect to my home network at the moment so can't start a test until tonight. I think my next test will be to launch the player from python to review behaviour.

riban-bw, 2019-05-23:
I ran this from the python interpreter using Popen (but not within a thread) for 9 hours without issue. I am now running from Zynthian to see if it triggers within next 9 hours.

riban-bw, 2019-05-23:
+9 hours and still going. Will leave it overnight...

riban-bw, 2019-05-24:
May 23 09:41:43 - From within zynthian, started player looping 4s MIDI file.
May 23 22:44:07 - Last logged "End Of Track".
May 24 01:13:11 - 3 XRUN messages (may not be relevant to this issue).

RPi was booted at 09:12:24.

So this test took 13:02:24 (46944s +4/-0) to trigger and triggered 13:31:43 (48703s +4/-0) after boot.

Next test is to run jack-smf-player for longer than 13 hours. First I will update to Aruk RC2.

riban-bw, 2019-05-25:
This issue occurred with jack-smf-player running standalone. I started it from the command line and some hours later it stopped logging "End Of Track" and no more MIDI input to Fluidsynth. This was still feeding ZynMidiRouter so to be absolutely sure it is not zynthian related we should run it without zynthian running but it does look like an upstream issue. There is no error message when it stops and I don't see an option to increase verbosity of jack-smf-player.

mheidt, 2019-12-22:
@jofemodo 
I vote for evaluating a different midi player.
I have a simple C4 4 beat measure mid-file that I want to play in a loop.
jack-smf-player can't do it on beat...

(on midi channel 11)

[C4th.mid.zip](https://github.com/zynthian/zynthian-ui/files/3992671/C4th.mid.zip)

riban-bw, 2021-07-05:
Zynthian now has bespoke SMF player.

closed 2021-07-05 by riban-bw