zynthian/zynthian-ui #126: Segmentation fault when exiting Zynthian

jofemodo, 2019-05-23:
Random Segmentation Faults are raised when exiting zynthian-ui. It seems to be caused for threads not being ended in a proper way. The problem has been there for a long time, but hidden. With the status bar implementation, has raised to the surface.

mheidt, 2019-05-23:
> Closed  #126 via  0d676f7 .

You didn't mean to close it,  right?

jofemodo, 2019-05-23:
No no ... the problem is still alive ;-(

jofemodo, 2019-05-30:
OK! I finally caught where the problem is. The jackpeak's "endJackpeak" function that is registered as "atexit" function is causing the problem. Disabling it solves the problem.

closed 2019-05-30 by jofemodo

riban-bw, 2019-05-30:
Let me take a look at that. I inherited that bit of code and didn't review it to closely. Maybe I should.

riban-bw, 2019-05-31:
If you remove the atexit() registration then you should also remove endJackpeak() function which is now obsolete. I wonder though whether the randomness of this is dependant on whether initJackpeak() succeeds. If init fails to create the client then endJackpeak will seg fault because g_pJackClient has not been assigned. This should be caught by:
```
void endJackpeak() {
	if(g_pJackClient)
		jack_client_close(g_pJackClient);
}
```
I am not sure that matches the bug report though. The issue reported relates to zunthian-ui exit at which point one would not expect jackpeak to be initialising.

Is there a test scenario which (reliably) triggers this issue? I want to be sure that the fix (not closing the JACK client) does not have any adverse affect.

commit 0d676f78 (2019-05-23): Fix #126: Random segmentation faults when exiting zynthian. Waiting for threads to end solves the problem.
commit 26e5b78f (2019-05-23): Fix #126: Segmentation fault when exiting zynthian-ui. Await zynautoconnect thread to end. Delay OSC starting and anticipate OSC ending.
commit 698bd592 (2019-05-30): Fix #126: Disable atexit function in jackpeak.
It was causing the problem from the beggining, but it served to improve exit code ;-)